contiki-os/contiki #2366: simple LED and printf on cc2650 launchpad

antigravity99, 2017-10-27:
I am completely new to contiki and embedded development (just a little experience with Arduino).

I cloned contiki on an Ubuntu 17.10 VM and can compile and run simple applications to make an LED blink for a TelosB sky mote and I am trying to get the same simple application to run for the cc2650 Launchpad. It compiles just fine and the led turns on with the first 'leds_toggle(CC2650_RED_LED);' but never toggles off, which tells me something is going wrong but I have no idea as to what and 'printf' never outputs anything. I have tried using minicom connecting to both /dev/ttyACM0 and /dev/ttyACM1 but I get no printf serial output.

Here is the code:

```
#include "contiki.h"
#include "sys/etimer.h"
#include "dev/leds.h"
#include <stdio.h>

#define CC2650_RED_LED       LEDS_RED

PROCESS(test_process, "test process");
AUTOSTART_PROCESSES(&test_process);
PROCESS_THREAD(test_process, ev, data)
{
  	PROCESS_BEGIN();

    leds_toggle(CC2650_RED_LED);

    static struct etimer et;

    etimer_set(&et, CLOCK_SECOND * 2);

    static int i = 1;
    static clock_time_t clockTime;

    while(i <= 100 )
    {
      etimer_set(&et, CLOCK_SECOND * 2);

      PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&et));

      clockTime = clock_time();

      printf("\rLoop %d     Clock Time %d", i++, (int)clockTime);
      leds_toggle(CC2650_RED_LED);

    }
    PROCESS_END();
}
```
I compile it with this command:
'sudo make clean TARGET=srf06-cc26xx BOARD=launchpad/cc2650 simple-test'

Again, this works just fine on a Sky mote, but the red led turns on and never turns off on the cc2650 launchpad and there is not printf serial output.

Any help will be greatly appreciated!

chenek, 2017-11-06:
Try to refer to http://sunmaysky.blogspot.tw/2017/10/using-cc13xxcc26xx-running-contiki-os.html

xioabudian, 2017-11-06:
in platform/srf06-cc26xx fold,different board has different peripherals configuration file,named board.h,and you can find UART pins,eg BOARD_IOID_UART_RX,BOARD_IOID_UART_TX, and try it on with USBTTL

xioabudian, 2017-11-06:
```
etimer_set(&et,CLOCK_SECOND * 2);
while(true){
PROCESS_YIELD_UNTIL(ev==PROCESS_EVENT_TIMER);
if(etimer_expired(&et)){
leds_toggle(LEDS_RED);
etimer_reset(&et);
}

}
```