contiki-os/contiki #2370: Simulation  Time in Cooja

Rjlmota, 2017-11-02:
Hello, there

I wanted to get the simulation time (like it shows in the cooja interface) in my code. But I can't seem to find where it comes from. Can somebody help me? Thank you.

fzy-Line, 2018-01-22:
In contiki/platform/cooja/sys/clock.c, the clock_time() or clock_seconds() function can return the current system clock time.

rahav, 2022-11-08:
I believe both of those functions are the time since a node started and not the Cooja time, which will be the time since the simulation started. I'm interested in the OP question as well, was there an answer posted somewhere? Docs?

fuad686337, 2024-09-30:
You can follow the steps from this page. [LINK ](https://slogix.in/source-code/contiki-cooja-samples-for-IoT/how-to-change-the-simulation-time-in-contiki-cooja-simulator/)