contiki-os/contiki #2387: Problems with TSCH scheduling

BowenWang, 2017-11-09:
Hi,
I am trying to use the TSCH library alone(without 6Tisch stuff). I have written a very simple network layer only support unicast and broadcast. 
I use the function tsch_schedule_add_link() to add slots in the frame.

My schedule is pretty simple:
slot   source   destination
1       113        110
2       116        113
3        118        116

I have done a lot of experiments, the result is that only slot 1 works fine. Node 110 can receive data from 113 while other two slots don't work.

I have checked the code, inside the TSCH library, there is a rx_slot process will be run every time it is a receiving slot, on node 113 and 116, every time it runs into rx_slot, it first check RADIO_DRIVER.receiving_packet(), this always returns true, then it will sleep for some time, after wake up, node will check RADIO_DRIVER.pending_packet(). The difference is at this function, pending_packet() will return true on node 110, but it will return false on node 113 and 116.
Anyone knows how I can solve this problem?

alexrayne, 2017-11-14:
you should careful establish options on links RX and TX. are you check them?
pending_packet() - false when no packet are in air. this can be caused - sources are not sends.

BowenWang, 2017-11-23:
@alexrayne 
When I add schedule information, I use the function tsch_schedule_add_link() like this
If current node is trying to receive information on that slot
tsch_schedule_add_link(RX|TIME_KEEPING, NORMAL_LINK)
If current node is trying to send information on that slot
tsch_schedule_add_link(TX, NORMAL_LINK)

Is this correct?

alexrayne, 2017-11-23:
that`s correct.
also link`s should carefuly note adress of target. if target not mutch, ink will not send. broadcast adress can be used.
you can set RX | TX - this prefere sending on slot if something need to be send, and receives if nothing to send.
I not use TIME_KEEPING - it is not implemented, and even not known it`s purposes.
link`s builds once after association, at net startup.

you shoud check that your 116,118 are sending.