contiki-os/contiki #2390: Wrong PIC32 rtimer configuration for creator branch

myrrdyn, 2017-11-11:
The default rtimer configuration for pic32 uses timer 2/3, but on the creator branch this is changed to make use of timer 4/5. Unfortunately the wrong timer number is then used, preventing rtimer from working.
The attached patch solves the issue
[0001-cpu-pic32-Fix-use-of-timer-4-5-for-rtimer.patch.zip](https://github.com/contiki-os/contiki/files/1464113/0001-cpu-pic32-Fix-use-of-timer-4-5-for-rtimer.patch.zip)

alexrayne, 2017-12-07:
why you wan`t create pull request?

myrrdyn, 2017-12-09:
Sorry, by mistake I opened a new issue on the wrong project, my patch only applies to the modified contiki for the CI40-kit clicker board supported by the CreatorDev/contiki fork.
I'm closing this issue now.

closed 2017-12-09 by myrrdyn