contiki-os/contiki #2406: Setting up 6lbr on raspberry - Contiki

ARmohankumar, 2017-12-07:
Hi experts,
Am new to Contiki-OS.
my primary objective is to setup 6lowPAN mesh network on cc1350 custom board,
Devices in my hand : raspberry pi 2 , one launchXL-CC130 board and few custom cc1350 board.

Some one please suggest the way of establishing the mesh network with available devices.
Any suggestions is appreciated.

Thanks & Regards,
AR mohan

chenek, 2017-12-11:
Try to refer to http://sunmaysky.blogspot.tw/2015/09/contiki-subg-hz-6lowpan-on-cc1350.html

cuongdodinh, 2017-12-11:
Dear Chenek,
If I don't want to ue Raspi, I want to use one Master with cc1310 LaunchPad, and all node using chip CC1310, Is it ok? It still has mesh network function? or if we want to use RPL, we must use Raspi for 6lbr

chenek, 2017-12-11:
You can refer to http://sunmaysky.blogspot.tw/2017/03/spi-connection-between-launchxl.html?m=1

cuongdodinh, 2017-12-11:
Dear Chenek,

So that, if we want to use 6lbr border router, we must using internet network? in this example we must use ENC28J60, if we don't use ENC28J60, we can't create 6LowPan network. Is it true? I only want to one master and some node and setup a mesh network.
Because our device place outside without internet network or switch.

chenek, 2017-12-11:
Try to refer to http://sunmaysky.blogspot.tw/2017/03/how-to-run-contiki-udp-serverudp-client.html . UDP server/client examples don't need to run with 6lbr.

cuongdodinh, 2017-12-11:
Dear Chenek,

If we using udp-server/udp-client, It's still make mesh network in this method?, not a star network. Because I want to use mesh function in this our module.

Master<===> node1 <<=250m=> node2 <<=250m=>> node3 <=250m=>node4.....node end.

    <-----------------------------------------------3km--->5km----------------------------------------->

chenek, 2017-12-11:
As I know, Contiki OS support mesh by default so it should be no problem.

cuongdodinh, 2017-12-11:
Dear Chenek,

Thank you so much.

allianceshan, 2018-11-21:
hi chenek !
   sorry,  i find these links can't pen . so can you kindly post these urls again?
   thank you so much!