contiki-os/contiki #2410: cc26xx-client-demo make file error

cursedcelestial, 2017-12-11:
in spoerk implementation  exmples\cc26xx client and server examples, can't make a bin file getting fatal: Not a git repository: '../../../.git'
make: *** No rule to make target `cc26xx-server-demo.bin'.  Stop.

chenek, 2017-12-12:
Can you specify which cc26xx example you are trying to build binary and how you build it?

cursedcelestial, 2017-12-12:
Thanks for the response. The compilation error is solved but i have have been stuck on the same example cc26xx-ble-client-demo & cc26xx-ble-server-demo.

To elaborate on the issue it is directed in the readme to add the server ip in client.c which i am doing by converting the ble mac to ipv6 but the server is not responding to client icmp ping, client is stuck in echo loop and the communication is not proceeding.

Best Regards

chenek, 2017-12-12:
I don’t hear cc26xx-ble-client-demo & cc26xx-ble-server-demo. where do you get them?

iqbal-h, 2018-01-02:
@cursedcelestial Have you resolved the ping issue using ble-client and ble-server example?
I am unable to ping too!