contiki-os/contiki #2413: example-broadcast

MarCast, 2017-12-18:
Hi everyone

I am creating a communication between two cc13xx devices with the sample-broadcast. I wanted to change the code, turning on a led from one device to another but I can not transmit. He manages to turn it on for himself but does not transmit it. I attach the code here I hope someone can help me.

#include "contiki.h"
#include "net/rime/rime.h"
#include "random.h"
#include "dev/serial-line.h"
#include "dev/cc26xx-uart.h"

#include "dev/button-sensor.h"

#include "dev/leds.h"

#include <stdio.h>
/*---------------------------------------------------------------------------*/
PROCESS(example_broadcast_process, "Broadcast example");
AUTOSTART_PROCESSES(&example_broadcast_process);

char cmd[5];
/*---------------------------------------------------------------------------*/
static void
broadcast_recv(struct broadcast_conn *c, const linkaddr_t *from)
{

char *appdata;

    if(appdata = packetbuf_dataptr()) {
    if(appdata[0]==0x4F && appdata[1]==0x4E){
        leds_on(LEDS_RED);
        printf("Open", from->u8[0], from->u8[1], (char *)packetbuf_dataptr());
        }
", from->u8[0], from->u8[1], (char *)packetbuf_dataptr());
        }
    if(appdata[0]==0x4F && appdata[1]==0x4E && appdata[2]==0x76){
        leds_off(LEDS_RED);
        printf("Close", from->u8[0], from->u8[1], (char *)packetbuf_dataptr());
        }}
static const struct broadcast_callbacks broadcast_call = {broadcast_recv};
static struct broadcast_conn broadcast;
/*---------------------------------------------------------------------------*/
PROCESS_THREAD(example_broadcast_process, ev, data)
{
  static struct etimer et;

  PROCESS_EXITHANDLER(broadcast_close(&broadcast);)

  PROCESS_BEGIN();
  cc26xx_uart_set_input(serial_line_input_byte);

  broadcast_open(&broadcast, 129, &broadcast_call);

  while(1) {
      PROCESS_YIELD();

   if(ev == serial_line_event_message && data != NULL) {
      if(strcmp(data,"ON")==0) {
        for(int i=0;i<5;i++)com[i]=0x0;
          cmd[0]=0x4F; 
          cmd[1]=0x4E;
          leds_on(LEDS_RED);
        } else if(strcmp(data,"OFF")==0) {
          for(int i=0;i<5;i++)com[i]=0x0;
          cmd[0]=0x4F;
          cmd[1]=0x4E;
          leds_off(LEDS_RED);
    }   
   /* Delay 2-4 seconds */
    etimer_set(&et, CLOCK_SECOND * 4 + random_rand() % (CLOCK_SECOND * 4));

    PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&et));
    packetbuf_copyfrom("start", 6);
    broadcast_send(&broadcast);
    //printf("broadcast message sent\n");
  }

  PROCESS_END();
}

greg-king5, 2017-12-20:
Your code puts the same data (`0x4F` and `0x4E`) into `cmd[]`, for both states (on and off).  But then, it broadcasts the string `"start"` instead.
Your code doesn't listen for `"start"`.  Instead, it listens for the letters `"ON"` and `"ONv"`.  Therefore, the LED can't change.

I suggest that you broadcast the word that was sent over the serial line:
```C
#include <stdio.h>
#include <string.h>

#include "contiki.h"
#include "net/rime/rime.h"
#include "random.h"
#include "dev/serial-line.h"
#include "dev/cc26xx-uart.h"

#include "dev/button-sensor.h"
#include "dev/leds.h"
/*---------------------------------------------------------------------------*/
PROCESS(example_broadcast_process, "Broadcast example");
AUTOSTART_PROCESSES(&example_broadcast_process);
/*---------------------------------------------------------------------------*/
static void
broadcast_recv(struct broadcast_conn *c, const linkaddr_t *from)
{
  char *appdata;

  if((appdata = packetbuf_dataptr()) != NULL) {
    if(strcmp(appdata, "ON") == 0) {
      leds_on(LEDS_RED);
      printf("Open heard from %u.%u\n", from->u8[0], from->u8[1]);
    } else if(strcmp(appdata, "OFF") == 0) {
      leds_off(LEDS_RED);
      printf("Close heard from %u.%u\n", from->u8[0], from->u8[1]);
    }
  }
}

static const struct broadcast_callbacks broadcast_call = {broadcast_recv};
static struct broadcast_conn broadcast;
/*---------------------------------------------------------------------------*/
PROCESS_THREAD(example_broadcast_process, ev, data)
{
  static struct etimer et;

  PROCESS_EXITHANDLER(broadcast_close(&broadcast);)

  PROCESS_BEGIN();
  cc26xx_uart_set_input(serial_line_input_byte);
  broadcast_open(&broadcast, 129, &broadcast_call);

  while(1) {
    PROCESS_YIELD();
    if(ev == serial_line_event_message && data != NULL) {
      if(strcmp(data,"ON")==0) {
        leds_on(LEDS_RED);
      } else if(strcmp(data,"OFF")==0) {
        leds_off(LEDS_RED);
      }

      /* Delay 2-4 seconds */
      etimer_set(&et, CLOCK_SECOND * 2 + random_rand() % (CLOCK_SECOND * 2));
      PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&et));
      packetbuf_copyfrom(data, strlen(data) + 1);  /* (copy '\0' also) */
      broadcast_send(&broadcast);
      //printf("broadcast message sent\n");
    }
  }

  PROCESS_END();
}
```

MarCast, 2017-12-22:
Thanks @greg-king5, i test this code, it send but not receive to another device

alexrayne, 2017-12-25:
cause your code have no any notifiation about received event.
you should poll example_broadcast_process from broadcast_recv

MarCast, 2017-12-27:
How? @alexrayne

alexrayne, 2017-12-27:
look for sys/process.h: process_poll

MarCast, 2017-12-29:
Hi @alexrayne thanks for your support. I see the process.h code, but what should I change in this code?

MarCast, 2018-01-08:
I insert the PROCESS_POLLHANDLER(); after PROCESS_BEGIN(), but not function

MarCast, 2018-01-08:
@alexrayne

alexrayne, 2018-01-08:
You should use function process_poll(event listner process)
just look for this function usage in contiki sources or examples.
Also Doxygen help explain this function well.

MarCast, 2018-01-08:
Are you sure about this use of the process_poll @alexrayne ?

alexrayne, 2018-01-08:
Yes

MarCast, 2018-01-09:
I don't understand how to use this process_poll, can you help me?

alexrayne, 2018-01-09:
look for usage exampes in :
examples\sensniff\sensniff.c
examples\cc26xx\cc26xx-web-demo\mqtt-client.c
core\lib\sensors.c
core\sys\etimer.c
core\dev\serial-line.c
...
and many more other stuff in contiky sources.
just do same

MarCast, 2018-01-09:
/*---------------------------------------------------------------------------*/
PROCESS(example_broadcast_process, "Broadcast example");
AUTOSTART_PROCESSES(&example_broadcast_process);
/*---------------------------------------------------------------------------*/
static void
broadcast_recv(struct broadcast_conn *c, const linkaddr_t *from)
{
  char *appdata;

  if((appdata = packetbuf_dataptr()) != NULL) {
    if(strcmp(appdata, "ON") == 0) {
      leds_on(LEDS_RED);
      printf("Open heard from %u.%u\n", from->u8[0], from->u8[1]);
    } else if(strcmp(appdata, "OFF") == 0) {
      leds_off(LEDS_RED);
      printf("Close heard from %u.%u\n", from->u8[0], from->u8[1]);
    }
     process_poll(&example_broadcast_process);
  }
}

static const struct broadcast_callbacks broadcast_call = {broadcast_recv};
static struct broadcast_conn broadcast;
/*---------------------------------------------------------------------------*/
PROCESS_THREAD(example_broadcast_process, ev, data)
{
  static struct etimer et;

  PROCESS_EXITHANDLER(broadcast_close(&broadcast);)

  PROCESS_BEGIN();
  cc26xx_uart_set_input(serial_line_input_byte);
  broadcast_open(&broadcast, 129, &broadcast_call);

  while(1) {
    PROCESS_YIELD();
    if(ev == serial_line_event_message && data != NULL) {
      if(strcmp(data,"ON")==0) {
        leds_on(LEDS_RED);
      } else if(strcmp(data,"OFF")==0) {
        leds_off(LEDS_RED);
      }

      /* Delay 2-4 seconds */
      etimer_set(&et, CLOCK_SECOND * 2 + random_rand() % (CLOCK_SECOND * 2));
      PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&et));
      packetbuf_copyfrom(data, strlen(data) + 1);  /* (copy '\0' also) */
      broadcast_send(&broadcast);
      //printf("broadcast message sent\n");
    }
  }

  PROCESS_END();
}

I modify the code with process_poll but not function

alexrayne, 2018-01-09:
process_poll forces example_broadcast_process  to pass 
` PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&et));`
but 
below `
packetbuf_copyfrom(data, strlen(data) + 1);  /* (copy '\0' also) */
`
using data argument not safe. since PROCESS_WAIT_EVENT_UNTIL waits any event, and data value comes from last event. if you want utilise data argument from code before  - `if(ev == serial_line_event_message && data != NULL)`, you should use it before PROCESS_WAIT_EVENT_UNTIL. atfer PROCESS_WAIT_EVENT_UNTIL data value looses

also, it not obvious - what to do example_broadcast_process  cycle?
why you delay 2-4sec before send, not after send?

alexrayne, 2018-01-09:
if you shurely want delay 2-4sec before send, you have to save somewhere data argument contents, for PROCESS_WAIT_EVENT_UNTIL time.

MarCast, 2018-01-09:
Ok thanks, I followed your advice. I moved EVENT_UNTIL before the if (ev ==) and only works with an ON command. When I type OFF it does not work

alexrayne, 2018-01-09:
packetbuf is common shared resource on whole net stack. you can`t rely that
it is keep old data while EVENT_UNTIL, cause net interupts can use it too.
you should build packetbuf and send it with not interuptable code. better
save your data in some local buffer while EVENT_UNTIL.

MarCast, 2018-01-09:
I'm new to this Contiki OS, how can I do this?

alexrayne, 2018-01-09:
just declare buffer for your string, and save data arg to it before
WAIT_UNTIL.

MarCast, 2018-01-09:
You can show me the code, I still can not understand

MarCast, 2018-01-10:
@alexrayne can you show me the code? please

alexrayne, 2018-01-10:
`
 // declare some temporary buffer 
 static buf[nnn];

  /* Delay 2-4 seconds */
  etimer_set(&et, CLOCK_SECOND * 2 + random_rand() % (CLOCK_SECOND * 2));
  strcpy(buf, data);                    // <- save data content before delay
  PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&et));
  // <- use here data saved befre delay 
  packetbuf_copyfrom(buf, strlen(buf) + 1);  /* (copy '\0' also) */
  broadcast_send(&broadcast);
`

MarCast, 2018-01-10:
Thanks @alexrayne. I delete the delay and I modify the code: 

`static buf[nnn];
PROCESS_EXITHANDLER(broadcast_close(&broadcast);)
PROCESS_BEGIN();
cc26xx_uart_set_input(serial_line_input_byte);
broadcast_open(&broadcast, 129, &broadcast_call);

strcpy(buf, data);
while(1) {
PROCESS_YIELD();
if(ev == serial_line_event_message && data != NULL) {
if(strcmp(data,"ON")==0) {
leds_on(LEDS_RED);
}
packetbuf_copyfrom(buf, strlen(buf) + 1);  /* (copy '\0' also) */
broadcast_send(&broadcast);`

It's correct?

alexrayne, 2018-01-10:
event and data actual value are changes after PROCESS_YIELD.
so 'if(ev == serial_line_event_message && data != NULL)' operates with new arrived values.
you should save this values right where ones are checked.

also, better be some delay in a cycle. without that - it goes faast.

MarCast, 2018-01-10:
How can I save them?

MarCast, 2018-01-10:
I test the code with the modify:

`// declare some temporary buffer
static buf[100];

/* Delay 2-4 seconds /
etimer_set(&et, CLOCK_SECOND * 2 + random_rand() % (CLOCK_SECOND * 2));
strcpy(buf, data); // <- save data content before delay
PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&et));
// <- use here data saved befre delay
packetbuf_copyfrom(buf, strlen(buf) + 1); / (copy '\0' also) */
broadcast_send(&broadcast);`

But not works, not sent the data

greg-king5, 2018-01-10:
Oops, Alex is correct that the value in `data` is temporary.  When the PROCESS_EVENT_...() macro leaves the process, the pointer is lost.  I added a few things (a `continue;` statement and a command buffer).
[Polling isn't needed by this simple example.  The process doesn't handle receive events.
Only broadcast_recv() does it.]

```C
#include <stdio.h>
#include <string.h>

#include "contiki.h"
#include "net/rime/rime.h"
#include "random.h"
#include "dev/serial-line.h"
#include "dev/cc26xx-uart.h"

#include "dev/button-sensor.h"
#include "dev/leds.h"
/*---------------------------------------------------------------------------*/
PROCESS(example_broadcast_process, "Broadcast example");
AUTOSTART_PROCESSES(&example_broadcast_process);
/*---------------------------------------------------------------------------*/
static void
broadcast_recv(struct broadcast_conn *c, const linkaddr_t *from)
{
  char *appdata;

  if((appdata = packetbuf_dataptr()) != NULL) {
    if(strcmp(appdata, "ON") == 0) {
      leds_on(LEDS_RED);
      printf("Open heard from %u.%u\n", from->u8[0], from->u8[1]);
    } else if(strcmp(appdata, "OFF") == 0) {
      leds_off(LEDS_RED);
      printf("Close heard from %u.%u\n", from->u8[0], from->u8[1]);
    }
  }
}

static const struct broadcast_callbacks broadcast_call = {broadcast_recv};
static struct broadcast_conn broadcast;
/*---------------------------------------------------------------------------*/
PROCESS_THREAD(example_broadcast_process, ev, data)
{
  static char command[3+1]; /* must be big enough for longest command string */
  static struct etimer et;

  PROCESS_EXITHANDLER(broadcast_close(&broadcast);)

  PROCESS_BEGIN();
  cc26xx_uart_set_input(serial_line_input_byte);
  broadcast_open(&broadcast, 129, &broadcast_call);

  while(1) {
    PROCESS_YIELD();
    if(ev == serial_line_event_message && data != NULL) {
      if(strcmp(data,"ON")==0) {
        leds_on(LEDS_RED);
      } else if(strcmp(data,"OFF")==0) {
        leds_off(LEDS_RED);
      } else {
        continue;
      }
      strcpy(command, data); /* remember it for later */

      /* Delay 2-4 seconds */
      etimer_set(&et, CLOCK_SECOND * 2 + random_rand() % (CLOCK_SECOND * 2));
      PROCESS_WAIT_EVENT_UNTIL(etimer_expired(&et));

      packetbuf_copyfrom(command, strlen(command) + 1);  /* (copy '\0' also) */
      broadcast_send(&broadcast);
      //printf("broadcast message sent\n");
    }
  }

  PROCESS_END();
}
```

greg-king5, 2018-01-10:
@alexrayne, an observer wants to see the LED change on both nodes.  They might be far from each other.  That observer needs time to go from the sender to the receiver; therefore, the first node delays before sending.

MarCast, 2018-01-10:
Hi @greg-king5 , I try your code but not receive nothing

alexrayne, 2018-01-10:
>  I try your code: when digit ON the led is on when digit OFF the led not off
is there some output in console?

MarCast, 2018-01-10:
@greg-king5  @alexrayne  Without the process_poll it does not receive anything, when I enter process_poll it works. But I keep sending only ON, OFF does not send it

alexrayne, 2018-01-10:
You better to add more printf messages, to see where you app is going

MarCast, 2018-01-10:
I can only turn on the led, when I type off the led of the first device goes off and sends the command, but the second one does not receive

alexrayne, 2018-01-10:
did the led-on command receives by remote?
if you dump received message  what it shows?

MarCast, 2018-01-10:
[image]

This is my problem

greg-king5, 2018-01-10:
I think that I misunderstood the comments for `packetbuf_clear()` (I had thought that it's automatic).
Put this line just above the `packetbuf_copyfrom` line:
```C
packetbuf_clear();
```

alexrayne, 2018-01-11:
packetbuf_clear helps?

MarCast, 2018-01-11:
Thanks for your supports! It's works! 

But this example, can i add to folder of cc26xx-web-demo?

MarCast, 2018-01-11:
@alexrayne @greg-king5

MarCast, 2018-01-15:
@alexrayne  It's possible to add this example with the folder of cc26xx-web-demo?

closed 2018-01-16 by MarCast