contiki-os/contiki #2415: Rpl-Tsch: Error while creating cooja mote

Siddarth1511, 2017-12-25:
Hi 
I am new to using contiki/cooja, so i was going around executing the example programs. My task was to evaluate Rpl-tsch in cooja as a beginning task for my final semester project. When i downloaded RPL-TSCH package from github and compiled it to create a cooja mote. it shows " Error when creating Corecomm instance: Lib1 (photos attached below). What should i be doing

[image]
[image]

Siddarth1511, 2017-12-25:
Can anyone please respond to this as this is important to me. Thank you in advance.

lamnd09, 2018-01-03:
It works well with Z1 node, maybe you can try it.

Siddarth1511, 2018-01-03:
@lamnd09 
Tried with Z1 mote,
says data will not fit in 'rom'
data overflow by 2 bytes

Siddarth1511, 2018-01-03:
what should i do

lamnd09, 2018-01-03:
Hello, 
 I have worked with RPL-TSCH network and tried Z1, (I guess that you run rpl-tsch example). 
And you need to upgrade msp430 version to 4.7.3. 
Please refer thread #2280  , this thread have detail information.

Siddarth1511, 2018-01-03:
okay 
I will try it

Siddarth1511, 2018-01-10:
@lamnd09 
hello,
i don't understand the instructions given in README for upgrading msp430-gcc 4.7.3(https://github.com/pksec/msp430-gcc-4.7.3)

Can you please help me with the steps and tell me in a simplified manner. I am new to this
Thank you in advance

lamnd09, 2018-01-11:
Hello, 
did you follow the instruction and run commands ? 
this instruction is just used to install msp430-4..7.3

Siddarth1511, 2018-01-11:
@lamnd09 
hello,
I am not able to follow the instructions
They are vague for me.
Can you please tell me how to do it
Please

lamnd09, 2018-01-11:
Hi, 
So, please follow these steps: 
step 1: Download file mspgcc-4.7.3.tar.bz2 and extract the tar in any location, then copy it to /opt/ 
step 2:  open HOME/.bashrc by using command "vi ~/.bashrc", and add the line "export PATH=$PATH:/opt/compilers/mspgcc-4.7.3/bin" to the end of bashrc file. 
Or you can run  "export PATH=$PATH:/opt/mspgcc-4.7.3/bin" directly from command line. 

Finally, you can use cooja to compile your code using upgraded msp version.

Siddarth1511, 2018-01-11:
@lamnd09 
It is not allowing me to copy to /opt/
how to grant the access

Siddarth1511, 2018-01-11:
@lamnd09 
Hi,

I have done the two steps that you said
after running "export PATH=$PATH:/opt/mspgcc-4.7.3/bin" directly in terminal it doesn't show me anything 
Does it mean i have added it?

Siddarth1511, 2018-01-11:
file:///home/user/Pictures/Screenshot%20from%202018-01-11%2012:26:39.png
file:///home/user/Pictures/Screenshot%20from%202018-01-11%2012:24:39.png

i added to the end of the file 
and gave the command 
source ~/.bashrc (for effect of change to take place)

still shows me the same error "data will not fit in 'rom' "

when i checked for the version using msp430-gcc --version)

what did i miss?

fzy-Line, 2018-01-22:
you can try the Contiki-ng: https://github.com/contiki-ng/contiki-ng.

Siddarth1511, 2018-04-03:
thank you so much for your help@lamnd09 @fzy-Line 
I tried a different approach
I took a different program modified the project-conf.h and used it.

closed 2018-04-03 by Siddarth1511

Dawitdjn, 2018-05-29:
Hello,
i'm new to contiki/cooja. so any one please help me how to implement ECC and NTRU with contiki-ng.
and the steps i should follow.

Haneenh, 2019-01-07:
@Siddarth1511 
please, can you tell me how you fixed above problem? 
I did all steps as you mentioned:
1- copy mspgcc-4.7.3 to /opt path
2- Add PATH MSP430/bin path to my environment using "export PATH=$PATH:/opt/mspgcc-4.7.3/bin"  command, then use source ~/.bashrc to take updates.
3- add "export PATH=$PATH:/opt/compilers/mspgcc-4.7.3/bin" command at the end of .bashrc file.

But when I check from version is still 4.7.0! 
Any suggestions?