contiki-os/contiki #2416: How to modify Flatform for CC1310 owner board.

cuongdodinh, 2017-12-29:
Hi All,

I make a board use CC1310F128RHB, this board has one LED connect to DIO3,
I modifie in contiki/platform/srf06-cc26xx/launchpad
#define BOARD_IOID_LED_1          IOID_3
I use example rpl-udp in /ipv6,
after make bin file, and use flash 2 to download firmware, but it's not run.
I'm not sure I do right.

Please show me the way how can I do if we build a circuit, not use LaunchPad CC1310.

Thanks,
Cuong

chenek, 2018-01-02:
The original board file is for 7x7 package and you should revise it to only use pins of 5x5 package.

cuongdodinh, 2018-01-03:
Dear @chenek ,

Thank for your help. It works OK. I want to get LQI, and RSSI in package, show I call funtion:

aux = packetbuf_attr(PACKETBUF_ATTR_RSSI);
  printf("RSSI: %ddBm ", (int8_t)aux);

  aux = packetbuf_attr(PACKETBUF_ATTR_LINK_QUALITY);
  printf("LQI: %u\n", aux);
DATA recv 'Reply' (s:101, r:101)
CH: 25 RSSI: -15dBm LQI: 0
DATA send to 1 'Hello 102'
DATA recv 'Reply' (s:102, r:102)
CH: 25 RSSI: -15dBm LQI: 0
DATA send to 1 'Hello 103'
DATA recv 'Reply' (s:103, r:103)
CH: 25 RSSI: -15dBm LQI: 0
But LQI allway = 0,

What happen? How can I fix it?

Thanks.

chenek, 2018-01-04:
You shall use sicslowpan_get_last_rssi() to get RSSI.

cuongdodinh, 2018-01-04:
Dear @chenek,

Thanks, How about LQI, If we use cooja to view simulation, LQI = 37, But in this I see a channel it shows = 26, In project-conf.h I modified to add row #define RF_CORE_CONF_CHANNEL 25
And file contiki-conf.h in platform, I changed #define RF_CORE_CONF_CHANNEL                     25 too.
What wrong?

Thanks,

chenek, 2018-01-04:
I simulation you can have LQI. However, I am not aware Contiki OS provide any LQI get API for CC13xx/CC26xx.

cuongdodinh, 2018-01-05:
Dear @chenek 

By the way, I want to ask you, How can I  Create /etc/6lbr/6lbr.conf file? It's show access denied or not permission?

chenek, 2018-01-05:
Use sudo to create it.

greg-king5, 2018-02-06:
> How about LQI?  If we use Cooja to view simulation, LQI = 37. But in this, I see a channel; it shows = 26. In "project-conf.h", I modified to add row
> ```c
> #define RF_CORE_CONF_CHANNEL 25
> ```
> And, in file "contiki-conf.h" in platform, I changed
> ```c
> #define RF_CORE_CONF_CHANNEL 25
> ```
> too. What wrong?

Don't touch `RF_CORE_CONF_CHANNEL`.  Instead,
```c
#define RF_CHANNEL 25
```
in "project-conf.h".