contiki-os/contiki #2420: contiki 2.7 -> 3.0 issues

avitzavi, 2018-01-08:
Hi,
I have recently updated to contiki 3.0 from 2.7 and am having issues updating my code to work on 3.0.

Firstly, I noticed that I had to change all mentions of rimeaddr_t to linkaddr_t, is this change mentioned anywhere? its not in the changelog.

Now I'm stuck with errors akin to:
Undefined reference to 'broadcast_send'
Undefined reference to 'broadcast_close'
etc.

Can anyone help me out with this?

greg-king5, 2018-01-10:
There is the line
```make
include $(CONTIKI)/Makefile.include
```
at the bottom of your Makefiles.  You need to put the line
```make
CONTIKI_WITH_RIME = 1
```
just above it.  That line tells "Makefile.include" to put Rime functions into the Contiki library that's linked into your programs.

(Look at the examples in "examples/rime/".)

avitzavi, 2018-01-10:
This solved my problem! 
Is there a changelog or some documentation of things that are now required in 3.0?

closed 2018-01-10 by avitzavi

greg-king5, 2018-01-10:
I think that there aren't any "migration" documents in the Contiki project.