contiki-os/contiki #2439: Mesh network code

pavankumar8279, 2018-02-05:
Hi 
I am trying to create mesh network using cc2650 launchpad five in number making four as nodes
and one board as gateway I am planning to use udp-client.c/udp-server.c example to send and receive data 
so how can I add mesh network features to it so that data will be received efficiently using ipv6 stack?
or Is there any other example code in Contiki except "rime"
or any functions available to use it for creating a mesh network

Thank you and Regards
Pavan

ankit2002, 2018-09-22:
Hey Pavan, Have you got any solution for this?

pavankumar8279, 2018-09-24:
Yes, the udp-client.c and udp-server.c code are itself creates a mesh network, that I didn't knew at that time and I tested on cc2650 launchpad

gnamyy, 2019-06-13:
Hi all,
sorry for intrusion
which is the command to compile the rpl-udp?