contiki-os/contiki #2446: 6LoWPAN MQTT cross demo

Sinopteek, 2018-02-25:
Hello everyone. 

We want to make the demo, that can be very usefull for some people starting learn Contiki.
See structure below. 
There are 6LBR run on [Beagle Bone Black](https://beagleboard.org/black) + 3 [Launchpads](http://www.ti.com/tool/LAUNCHXL-CC1310) (one as radio interface and 2 as end nodes). 
[image]

We want to made simple demo where when we pressed button on first launchpad it'll publish message on topic A, while second launchpad subscribed on this topic and light the LED. And backwards - Launchpad 2 publish on topic B when button will pressed, that should lead to turn LED on on Launchpad 1. 

We want to run MQTT broker (we using mosquitto) on BBB, because it's looks most simple way, but there we got some strange behavoir. 

1. We can't access to mosquitto via ipV6 from BBB (linux).
2. We can't ping, traceroute to local ipV6 inside BBB (only ipV4 or localhost)
3. We can start mosquitto on PC and can access from BBB and routers (6LoWPAN nodes) with MQTT demo, but:
4. We can publish from router when button pressed (we see it from external subsciber for debug),
5. We can publish from external application in the same topic, and LED will lught on on another Launchpad (so mqtt works on both launchpads in both directions).
6. But when a Launchpad publish in the topic, another one don't light LED on. 

We see in some topics that Linux have Kernel limitations on BBB. But without details, so we don't sure about the reasons and way of solutions. It's manifest only on BBB and we can just switch to another SOM (Raspberry PI for example)? Or we should run MQTT broker on external host anyway (it's looks very inconveniently for portable demo and small project)? 
What may be wrong with demos, that leads for behavoir we can't send data between Launchpads, while it work with external application at the same time?

lligios, 2018-02-26:
First of all 
Try to start Mosquitto AFTER 6lbr , when you are sure that TAP device is active.

Second:
use ping6 for IPV6 command instead of ping. 

Third:
Which mosquitto instance are Launchpad node subscribing when you see the correct behaviour?

From you list at the it looks like launchpad are publishing a not correct MQTT message..
be aware of extra characters and case sensitiveness.
Avoid if possibile wildcard subscription.. or just use one at the client side.

Sinopteek, 2018-02-26:
Hello lligios 
1. We'll try it and answer here the results.
2. Sure we use ping6 for ping of IPV6 addresses. But it's useless.
3. We can connect from launchpad nodes only to Mosquitto on PC. Some comments to clarify:
We can't connect from Launchpad nodes to Mosquitto on BBB (because we can't access BBB IPv6 from anywhwere). We can connect to Mosquitto on BBB from PC's client, but only via IPv4.

We publish just "1" or "0". We can read these messages from PC client as expected.
We don't use wildcards from Launchpad nodes. Only $SYS/# from PC client for debug.

Sinopteek, 2018-02-28:
I made one more scheme to clarify what's going wrong
[image]

jsponz, 2018-03-02:
Any update on this? I am interested.

dmkuzmin, 2018-03-04:
1. Routing between all devices configured on the link https://sunmaysky.blogspot.ru/2017/10/how-to-configure-6lbr-to-make-it-can-do.html. Parameter CREATE_BRIDGE is set to "1". The Mosquito is now set to BBB.

2. The problem with MQTT was that the clients themselves had the same client_id. Because of this, clients could take turns connecting to the broker.

closed 2018-03-06 by Sinopteek