contiki-os/contiki #2453: soc_trng_rand_synchronous failure handling

EricSesterhennX41, 2018-03-09:
soc_trng_rand_synchronous() in cpu/cc26xx-cc13xx/dev/soc-trng.c can fail and return 0, this failure is not made transparent to the caller, which might cause them to use bad entropy.

alexrayne, 2018-03-13:
What you can propose to solve this? 
maybe provide some function - pseudo-random generator based on last randoms hystory?

EricSesterhennX41, 2018-03-14:
I guess in a first step, make sure that the caller is able to get notified of the error by changing the API. eg int soc_trng_rand_synchronous(uint64_t &rand); where it returns a status value and requires a pointer to the random variable.
In the long term i would propose to add a general randomness infrastructure using randomness pools which are feed from various sources and able to almost always provide randomness, similar to /dev/[u]random on *nix.