contiki-os/contiki #2455: CC1310_Contiki_Mesh

rajesh9949, 2018-03-20:
Dear Sir,
please clarify the below querries:

 1)We are using the Chip set CC1310F128RSMR.we are working the latest contiki 3.0 from the Git.I am currently developing on the CC1310 and running Conitki on it for 6L0WPAN .when we test the nodes (slave's)inside the buildings the mesh is forming and we are able send & receive data via hops (we observed in sniffer wireshark) but the same nodes when we check line of sight(open sky) then the mesh is not forming & only star network is forming  as observed in the sniffer.please clarify my dought & explain the mesh scenario how much max distance the nodes will cover.

2)while compiling the project Baudrate of the module we are setting in "platform/srf06/contikicong.h". But I want to save the baudrate,Channel in some locations in CC1310 memory.

I want to configure the baudrate  & channel using PC serial port ( user will change as per the meter make).How can i save the baudrate & channel.i dont want to external flash.i want to use internal flash.

Rajesh.M

MartinNPN, 2018-04-24:
Hello Rajesh,

regarding your first question, i don't know your exact hardware, but if they are outside and have good a good rf connection, this is normal, because they calculate the best way to the border router and if it is a direct connection they will use it.

Regards,
Martin

rajesh9949, 2018-04-27:
Dear sir,

Can you please answer the below querries.

1)Thanks you sir, we are using udpserver for master & udpclient for node side.That will create any mesh problem.

2)while compiling the project Baudrate of the module we are setting in "platform/srf06/contikicong.h". But I want to save the baudrate,Channel in some locations in CC1310 memory.

I want to configure the baudrate & channel using PC serial port ( user will change as per the meter make).How can i save the baudrate & channel.i dont want to external flash.i want to use internal flash.

3)Can you please provide the OTA example for the Chip set CC1310F128RSMR.

Please  reply to all

Rajesh.M