contiki-os/contiki #2460: tsch:! radio does not support getting last packet timestamp. abort init.

Siddarth1511, 2018-04-03:
After solving my previous issue #2415 .
I started working with rime stack along with tsch as mac layer. I tried to do the same thing i did using rpl-tsch by modifying the project-conf.h file.
Everything went well and i created the mote.
But when i run the simulation. It gives me this
"tsch:! radio does not support getting last packet timestamp. abort init" 
[image]

Siddarth1511, 2018-04-03:
Please help.

Siddarth1511, 2018-04-03:
Hi,

I solved the first error "TSCH:! radio does not support getting last packet timestamp. abort init."
This error occured because i did not include 
/* Disable DCO calibration (uses timerB) */
#undef DCOSYNCH_CONF_ENABLED
#define DCOSYNCH_CONF_ENABLED			 0

/* Enable SFD timestamps (uses timerB) */
#undef CC2420_CONF_SFD_TIMESTAMPS
#define CC2420_CONF_SFD_TIMESTAMPS       1
This is my project-conf.h file as Z1 mote uses cc2420 2.4 Ghz RF transmitter in it.

But now I got a new problem after that.

it says TSCH:! not associated, drop outgoing packet. and just scans the channel. it is not broadcasting the data
[image]

I face the similar problem that was put up in #1785 by @lamnd09

And he closed that issue.
If someone face/faced this issue
Please help me

lamnd09, 2018-04-04:
Hello, if you use Orchestra scheduling, maybe you need to enable it by changing number from 0 to 1in project-conf.h file.  It will work. 
/* Set to run orchestra */
#ifndef WITH_ORCHESTRA
#define WITH_ORCHESTRA 1
#endif /* WITH_ORCHESTRA */

by any chance, are there two nodes in a communication range ?

Siddarth1511, 2018-04-04:
Hi @lamnd09 , 
I am not using orchestra scheduling.  Should i use it.
And as you can see the picture in the previous comment. i have used two nodes. using DGRM radio environment.

lamnd09, 2018-04-04:
In my opinion, maybe you forgot to enable an timeslot scheduling in your simulation. 
If you do not propose a new scheduling algorithm and evaluate the performance, so Orchestra is good choice and  enabling Orchestra is not effect to your simulation.

lamnd09, 2018-04-04:
rpl-tsch is simple example of rpl-tsch network. First you need to decide one node to be root, and compile it separately, then compile others.

Siddarth1511, 2018-04-04:
Thank you.
i guess i did not include any time scheduling algorithm
I will include orchestra and let you know if i get the result.

And also i am not using rpl-tsch program.
I am trying to implement tsch in rime stack.
Although i will bring in rpl network stack to it eventually, so i will keep in mind what you just told.

Siddarth1511, 2018-04-04:
how do we fix a root node. 
should we add something in project-conf.h?

lamnd09, 2018-04-04:
Hi, 
In default settings,you do not need to configure root, it will automatically decide a root (node 1) or not, in my experience, it's not. 
If not, you maybe need to consider which root to be 6dr, and other 6ln, smt like that in the node.c file.

Siddarth1511, 2018-04-04:
alright, tell me if i understood this correctly.
By what you said, if i just compile one node separately, by default, it is the root node.
Am i right?
Or did i misinterpret it.

Siddarth1511, 2018-04-04:
okay great! 
i will work it out and let you know. 
Thank you so much for your help.

By the way, is there an example program for root.c ?

lamnd09, 2018-04-04:
the root.c is very similar with node.c, it is just different in mode (6dr), maybe you need to find the line of code which decide the working mode, I do not remember exactly, but it should be in this 
#define CONFIG_WAIT_TIME 5

    SENSORS_ACTIVATE(button_sensor);
    etimer_set(&et, CLOCK_SECOND * CONFIG_WAIT_TIME);

    while(!etimer_expired(&et)) {
      printf("Init: current role: %s. Will start in %u seconds. Press user button to toggle mode.\n",
             node_role == role_6ln ? "6ln" : (node_role == role_6dr) ? "6dr" : "6dr-sec",
             CONFIG_WAIT_TIME);
      PROCESS_WAIT_EVENT_UNTIL(((ev == sensors_event) &&
                                (data == &button_sensor) && button_sensor.value(0) > 0)
                               || etimer_expired(&et));
      if(ev == sensors_event && data == &button_sensor && button_sensor.value(0) > 0) {
        node_role = (node_role + 1) % 3;
        if(LLSEC802154_ENABLED == 0 && node_role == role_6dr_sec) {
          node_role = (node_role + 1) % 3;
        }
        etimer_restart(&et);
      }
    }
  }

#endif /* CONFIG_VIA_BUTTON */

  printf("Init: node starting with role %s\n",
         node_role == role_6ln ? "6ln" : (node_role == role_6dr) ? "6dr" : "6dr-sec");

  tsch_set_pan_secured(LLSEC802154_ENABLED && (node_role == role_6dr_sec));
  is_coordinator = node_role > role_6ln;

  if(is_coordinator) {
    uip_ipaddr_t prefix;
    uip_ip6addr(&prefix, UIP_DS6_DEFAULT_PREFIX, 0, 0, 0, 0, 0, 0, 0);
    net_init(&prefix);
  } else {
    net_init(NULL);
  }

Siddarth1511, 2018-04-04:
Awesome!
I will first include orchestra scheduling in my program, then try to implement rpl with tsch as an extension and i will get back to you.

Thank you! @lamnd09

Siddarth1511, 2018-04-04:
I Solved it mate!
thanks for helping me out. ( not with rpl and tsch but with rime and tsch)

I set coordinator and destination address 
and used the function tsch_set_coordinator().

closed 2018-04-09 by Siddarth1511

souhilak, 2019-04-19:
@Siddarth1511 i have the same problem can you help me ?

MehdiKherb, 2020-03-22:
@Siddarth1511 hey mate, can you please tell us how you solved your issue, did you make two different code files, one for the coordinator and one for the simple node ? i'm trying to work on tsch with rime stack and i'm kinda stuck since the example provided in the contiki examples file doesn't work right for me.. it gives me the error : not associated : drop outgoing packet.

Siddarth1511, 2020-08-26:
@souhilak 
I am terribly sorry for such a late reply. I had closed the issue and project and did not come back to this. I guess you might be having options to assign the coordinator and destination address for the nodes in the options.

Siddarth1511, 2020-08-26:
@MehdiKherb 
I used the example codes provided with contiki, If I remember correctly, I used separate files for coordinator and other nodes. And I assigned them their respective addresses for broadcasting.