contiki-os/contiki #2461: unable to compile the server and client examples of tinydtls

nsainelson, 2018-04-05:
nelson@nelson-Aspire-E1-571:~/contiki1/apps/tinydtls$ make
make -C examples/contiki all
make[1]: Entering directory '/home/nelson/contiki1/apps/tinydtls/examples/contiki'
TARGET not defined, using target 'native'
mkdir obj_native
  CC        ../../../../apps/tinydtls/dtls.c
../../../../apps/tinydtls/dtls.c: In function ‘dtls_update_parameters’:
../../../../apps/tinydtls/dtls.c:908:3: warning: implicit declaration of function ‘assert’; did you mean ‘qsort’? [-Wimplicit-function-declaration]
   assert(config);
   ^~~~~~
   qsort
  CC        ../../../../apps/tinydtls/crypto.c
  CC        ../../../../apps/tinydtls/hmac.c
  CC        ../../../../apps/tinydtls/aes/rijndael.c
  CC        ../../../../apps/tinydtls/sha2/sha2.c
../../../../apps/tinydtls/sha2/sha2.c:42:2: warning: #warning "assertions are disabled" [-Wcpp]
 #warning "assertions are disabled"
  ^~~~~~~
  CC        ../../../../apps/tinydtls/ccm.c
../../../../apps/tinydtls/ccm.c: In function ‘add_auth_data’:
../../../../apps/tinydtls/ccm.c:117:3: warning: implicit declaration of function ‘assert’; did you mean ‘qsort’? [-Wimplicit-function-declaration]
   assert(la < 0xFF00);
   ^~~~~~
   qsort
  CC        ../../../../apps/tinydtls/netq.c
../../../../apps/tinydtls/netq.c:17:2: warning: #warning "assertions are disabled" [-Wcpp]
 #warning "assertions are disabled"
  ^~~~~~~
  CC        ../../../../apps/tinydtls/ecc/ecc.c
  CC        ../../../../apps/tinydtls/dtls_time.c
  CC        ../../../../apps/tinydtls/peer.c
  CC        ../../../../apps/tinydtls/session.c
../../../../apps/tinydtls/session.c:33:2: warning: #warning "assertions are disabled" [-Wcpp]
 #warning "assertions are disabled"
  ^~~~~~~
  CC        ../../../../apps/tinydtls/debug.c
../../../../apps/tinydtls/debug.c: In function ‘dsrv_print_addr’:
../../../../apps/tinydtls/debug.c:187:5: warning: #warning "IPv4 network addresses will not be included in debug output" [-Wcpp]
 #   warning "IPv4 network addresses will not be included in debug output"
     ^~~~~~~
  CC        ../../../../core/ctk/ctk-conio.c
  CC        ../../../../platform/native/./contiki-main.c
  CC        ../../../../platform/native/./clock.c
  CC        ../../../../core/dev/leds.c
  CC        ../../../../platform/native/dev/leds-arch.c
  CC        ../../../../platform/native/dev/button-sensor.c
  CC        ../../../../platform/native/dev/pir-sensor.c
  CC        ../../../../platform/native/dev/vib-sensor.c
  CC        ../../../../platform/native/dev/xmem.c
  CC        ../../../../core/lib/sensors.c
  CC        ../../../../platform/native/dev/irq.c
  CC        ../../../../core/cfs/cfs-posix.c
  CC        ../../../../core/cfs/cfs-posix-dir.c
  CC        ../../../../platform/native/ctk/ctk-curses.c
  CC        ../../../../cpu/native/net/tapdev-drv.c
  CC        ../../../../cpu/native/net/linuxradio-drv.c
  CC        ../../../../cpu/native/net/tapdev6.c
  CC        ../../../../cpu/native/./mtarch.c
  CC        ../../../../cpu/native/./rtimer-arch.c
  CC        ../../../../core/loader/elfloader-stub.c
  CC        ../../../../cpu/native/./watchdog.c
  CC        ../../../../cpu/native/dev/eeprom.c
  CC        ../../../../core/sys/autostart.c
  CC        ../../../../core/sys/ctimer.c
  CC        ../../../../core/sys/arg.c
  CC        ../../../../core/sys/energest.c
  CC        ../../../../core/sys/etimer.c
  CC        ../../../../core/sys/stimer.c
  CC        ../../../../core/sys/compower.c
  CC        ../../../../core/sys/process.c
[... 89 lines cut ...]
  AR        contiki-native.a
  CC        dtls-server.c
  LD        dtls-server.native
contiki-native.a(dtls.o): In function `dtls_update_parameters':
/home/nelson/contiki1/apps/tinydtls/examples/contiki/../../../../apps/tinydtls/dtls.c:908: undefined reference to `assert'
/home/nelson/contiki1/apps/tinydtls/examples/contiki/../../../../apps/tinydtls/dtls.c:909: undefined reference to `assert'
contiki-native.a(dtls.o): In function `clear_hs_hash':
/home/nelson/contiki1/apps/tinydtls/examples/contiki/../../../../apps/tinydtls/dtls.c:1088: undefined reference to `assert'
contiki-native.a(dtls.o): In function `dtls_verify_peer':
/home/nelson/contiki1/apps/tinydtls/examples/contiki/../../../../apps/tinydtls/dtls.c:1560: undefined reference to `assert'
/home/nelson/contiki1/apps/tinydtls/examples/contiki/../../../../apps/tinydtls/dtls.c:1592: undefined reference to `assert'
contiki-native.a(dtls.o):/home/nelson/contiki1/apps/tinydtls/examples/contiki/../../../../apps/tinydtls/dtls.c:1702: more undefined references to `assert' follow
contiki-native.a(dtls_time.o): In function `dtls_clock_init':
/home/nelson/contiki1/apps/tinydtls/examples/contiki/../../../../apps/tinydtls/dtls_time.c:40: undefined reference to `clock_init'
collect2: error: ld returned 1 exit status
../../../../Makefile.include:280: recipe for target 'dtls-server.native' failed
make[1]: *** [dtls-server.native] Error 1
rm dtls-server.co
make[1]: Leaving directory '/home/nelson/contiki1/apps/tinydtls/examples/contiki'
Makefile:98: recipe for target 'all' failed
make: *** [all] Error 2

make[1]: Leaving directory '/home/nelson/contiki1/apps/tinydtls/examples/contiki'
Makefile:98: recipe for target 'all' failed
make: *** [all] Error 2

slj2606, 2018-07-22:
Hello,
Was this issue fixed? if yes, please let me know how as I am getting the same error when I am trying to run the make for tinydtls in Contiki.

nusrathmouri, 2018-12-11:
Hi,
I am getting the same errors. Can anyone help to solve these issues?