contiki-os/contiki #2468: LLSec with hardware assisted CCM* for cc26xx/cc13xx

ram95014, 2018-04-17:
Is there somebody who can enable llsec on cc26xx platform using hardware assisted CCM*?

alexrayne, 2018-04-18:
see PR#2379

ram95014, 2018-04-18:
Alexander, this is great. I also have a working version of the AES/CCM*
implementation working. I should have checked your PR before I began work.
However, I took a different approach: I added crypto definitions in
ti_lib.h with macros from crypto.h in cc26xxware/driverlib. I have the
crypto working in standalone (tested with examples/llsec/ccm-star-tests),
but when I try it with NONCORESEC enabled, cc26xx-web-demo in combination
with slip-radio and a RPI native-border-router which works without crypto
does not work. Do you have a working version of NONCORESEC enabled
cc26xx-web-demo?

Also, are we waiting for adaptivesec (PR #1703) to come in before we can
use LLSEC on CC2650?

alexrayne, 2018-04-19:
My CCM implementation have ported like cc2538. since that one was inroduced by Konrad Krentz (maybe i miss?), i guess it should implements NONCORESEC well

As for AKES implentation by PR#1703. for my project that was sounds as overkill. therefore i have implements similar functionality by PR#2382 - this patch a bit enhances native TSCH stack, without introducing much more code, and no need additional LLSEC layer.
as fork AKES - it is interesting for implements secured broadcasts, but for trivial secure schemes, imho, it is overcomplicated.