contiki-os/contiki #2479: Cooja: msp code watcher cannot find source file

JunyangShi, 2018-05-25:
Hi,
I am trying to use cooja's msp code watcher to debug my code. But I go into the code watcher for a sky mote in Cooja, the code watcher displays: [no source diplayed]. So I click the "locate the source file" button. But still it does not match any source file. The error is listed below:

[/cygdrive/c/|c:/] does not match: /build/buildd/gcc-msp430-4.6.3~mspgcc-20120406/./gcc-4.6.3/libgcc/../gcc/config/msp430/libgcc.S.
[/cygdrive/c/|c:/] does not match: ./string/strncmp.c.
[/cygdrive/c/|c:/] does not match: ./string/memcmp.c
.....
.....
[/cygdrive/c/|c:/] does not match: /build/buildd/gcc-msp430-4.6.3~mspgcc-20120406/./gcc-4.6.3/gcc/config/msp430/crt0.S

Anyone who can help me with it? Thanks!

yrawna, 2019-02-16:
solved the problem for all files
only the first and last error not solved
/build/buildd/gcc...................

anyone can help????

junyang28, 2019-02-16:
Hi yrawna,
Could you tell me how you solve this kind of problem (except the first and last error)? Thanks.

yrawna, 2019-02-17:
located all source files that matches successfully
but
unfortunately MSP_CODE_WATCHER is not working properly.

search the files stncmp.c memcmp.c..............in Files or on the internet
then make a new directory name it as string in contiki folder
/home/user/contiki/string/
/home/user/contiki/stlib/
paste the corresponding files there.

in Cooja MSP_Code_Watcher window
click locate source files

then in Prefix column type
./
then in Substitution column type
/home/user/contiki/

do this for first and last error as well.

Apply
Save as default
Close

yrawna, 2019-02-25:
downloaded instant contiki3.x Virtual Machine.
MSP_CODE_WATCHER started to work fine

junyang28, 2019-02-25:
Hi yrawna,

Thanks! it works.