contiki-os/contiki #2495: Zolertia zoul dht22 example not working

orlax, 2018-07-13:
Hello, I am working with the zolertia soul re-mote revb, with the last version of the repo the example for getting readings from the dht22 temperature and humidity sensor is not working. 

contiki/examples/zolertia/zoul/test-dht22.c

after uploading it to the board and having the sensor connected it just says "could not connect to the sensor" 

I have try this same example with another repository: https://github.com/contiki-ng

and the exact same codes works as expeted and shows the lectures from the sensor, the problem is that contiki-ng doesn't has support for the rime library for radio communication. 

I think this must be something related to the sensor libraries on the repo? maybe a bad port configuration somewhere? it seems odd to me that the same code would work when compiling with contiki-ng but not with contiki alone. 

to replicate my issue: 

- clone the contiki repository
- install the submodules 
- flash the zolertia/zoul dht22 example to a remote rev-b board. 
- sudo make login to the board. 
- see the serial readings.

orlax, 2018-11-01:
thank you for getting to this, i have since move on from the project i was working on at the time. However will try to apply those changes and see about sending a pull request.

Akash25595, 2018-12-28:
Hi Brijesh,
Even i am also getting the error as failed to read the sensor for dht22 sensor. I have made the changes according to your last comment but values for temp and hum is displaying as 0.0 & 0.0 always.

brijesh027, 2019-01-15:
Hi Akash,
As per my experience it could mean either the Sensor is not receiving
proper power supply or the Sensor has gone bad. It could also mean that
proper pull has not been given to the data pin. To do a preliminary
examination to check if the Sensor is working you can run the code and
observe the data pin on an Oscilloscope.

akash2559, 2019-01-16:
Hi Brijesh,
Thanks for your mail. After making the multiple changes into the code of dht22.c i was able to run the code and see the proper output.

dserbu, 2019-03-16:
Hi @akash2559 , @brijesh027 . Which are exactly the modifications that need to be done, in order to see the temperature and humidity?

Akash25595, 2019-03-20:
Hi @dserbu , please make the changes accordingly as mentione bu @brijesh027 above, you will get the proper readings from DHT22 sensor.

dserbu, 2019-03-20:
> Hi @dserbu , please make the changes accordingly as mentione bu @brijesh027 above, you will get the proper readings from DHT22 sensor.

Done, thank you