contiki-os/contiki #2498: graph plot of cooja simulation output

Dipen404, 2018-07-18:
How can i plot graph from simulation output such as OF0 vs ETX. I also need to plot graphs related to energy consumption,packet delivery ratio,network convergence time,latency,hop count

lamnd09, 2018-07-24:
Hi, 
Obviously, you can collect output from Cooja and plot the graph. 
On other metrics, it's up to program/application running on your nodes.

lamnd09, 2018-07-24:
Hi, 
On energy consumption, you can use powertrace to measure it as instructions followed this link: http://thingschat.blogspot.com/2015/04/contiki-os-using-powertrace-and.html

PDR, Latency, hopcount, it's up to your application, read the output from cooja, export as text file, then use a python or perl script to read as casual.

Dipen404, 2018-07-24:
@lamnd09 thank you. But i don't know about perl script. It would be thankful if you can guide me to read the text files using perl script to plot graph

dharejo, 2019-10-19:
hi, i am just taking 10 motes of same type and testing their graph.
em not modifying code just using cooja example shell.c for simulation. but getting 0.0 latency graph can one help me how to plot graph of latency. 

[image]

bouktina, 2023-03-30:
* Can you guide me to read text files with perl script in cooja simulator 
* But I don't know anything about Perl script. I would be grateful for help