contiki-os/contiki #2505: perl script for mote output

Dipen404, 2018-08-24:
can anyone help me with the perl script to calculate energy consumption, latency, PDR, convergence time, control traffic overhead from the output of cooja simulation running RPL-UDP application

abbasalouswy, 2018-08-30:
hello all
I need same help for  Dipen404

erfanlinman, 2018-09-07:
.@Dipen404  look for the delay, PDR you could use the collect-view but for the energy, you need some change. additionally, I  advice you read the complete wiki in GitHub and Contiki 2.6 documents.

hieu1204, 2018-09-25:
Hi, I think this would be helpful
https://pdfs.semanticscholar.org/d319/4d5f43f1cd3995ba7c9e99776ac8f150d445.pdf

dharejo, 2019-10-19:
hi,
can anyone please help me how to find latency of sky motes, t-motes and wismotes em trying a lot but didn't find any tutorial related to my work

dharejo, 2019-10-19:
i hve performed simulation on these motes but found 0.0 value at latency graph so plz help me how plot graph of latency or calculate latency of motes in cooja.

sousouk, 2023-04-01:
Hello can you tell me  how I do script perl in cobtkiki cooja simulator  please help me