contiki-os/contiki #2506: mote output problem

Dipen404, 2018-08-29:
How can we know which data represents which value in cooja simulation mote output

erfanlinman, 2018-09-05:
could you describe more distinctly? 
you mean how we could print data in mote output tool?

Dipen404, 2018-09-06:
Cooja simulation mote output will generates some numerical values. The thing is it does not specify which data values is for which mote property like latency, energy consumption, PDR etc. I have read we can extract mote output in text files but if we don't know which row or column represent which property of RPL (ETX or hop count) or mote then how can we further process this with scripting language like perl script

erfanlinman, 2018-09-07:
I thought you are using the sink and sender module, as result you must read those modules to understood what is printing in the mote's output. And if you looking for the modules which their output are readable for the human you could use UDP client and server. And for those parameters, you want to measure them you must add some change in the output of your net part and app part code.

Dipen404, 2018-09-08:
@erfanlinman thank you but what changes should i make in net and app part code to to measure them

Dipen404, 2018-09-11:
@erfanlinman  thank you.