contiki-os/contiki #2509: ant run not opening cooja.showing following commands

fabijas, 2018-08-31:
at java.lang.Class.getDeclaredConstructors0(Native Method)
     [java] 	at java.lang.Class.privateGetDeclaredConstructors(Class.java:2671)
     [java] 	at java.lang.Class.getConstructor0(Class.java:3075)
     [java] 	at java.lang.Class.getConstructor(Class.java:1825)
     [java] 	at org.contikios.cooja.Cooja.registerPlugin(Cooja.java:1930)
     [java] 	at org.contikios.cooja.Cooja.reparseProjectConfig(Cooja.java:1576)
     [java] 	at org.contikios.cooja.Cooja.<init>(Cooja.java:434)
     [java] 	at org.contikios.cooja.Cooja$30.run(Cooja.java:3322)
     [java] 	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
     [java] 	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
     [java] 	at java.awt.EventQueue.access$500(EventQueue.java:97)
     [java] 	at java.awt.EventQueue$3.run(EventQueue.java:709)
     [java] 	at java.awt.EventQueue$3.run(EventQueue.java:703)
     [java] 	at java.security.AccessController.doPrivileged(Native Method)
     [java] 	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
     [java] 	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
     [java] 	at org.GNOME.Accessibility.AtkWrapper$6.dispatchEvent(AtkWrapper.java:715)
     [java] 	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
     [java] 	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
     [java] 	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
     [java] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
     [java] 	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
     [java] 	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

erfanlinman, 2018-09-05:
the data, which you provided here, is very low to could help you. could you describe more and clearly?

fabijas, 2018-09-05:
when I added mobility, After changing DEFAULT_PROJECTDIRs *to
[CONTIKI_DIR]/tools/cooja/apps/mobility*

I cant open the cooja simulator.when I run it shows following error

[delete] Deleting directory
/home/iiitmk/Downloads/contiki/contiki/tools/cooja/apps/serial_socket/build

BUILD FAILED
/home/iiitmk/Downloads/contiki/contiki/tools/cooja/build.xml:201: The
following error occurred while executing this line:
/home/iiitmk/Downloads/contiki/contiki/tools/cooja/apps/serial_socket/build.xml:22:
Unable to delete file
/home/iiitmk/Downloads/contiki/contiki/tools/cooja/apps/serial_socket/build/org/contikios/cooja/serialsocket/SerialSocketServer.class

erfanlinman, 2018-09-07:
Could you act like this [tutorial](http://anrg.usc.edu/contiki/index.php/Mobility_of_Nodes_in_Cooja) for mobility plugin.
 And by the way, are you using the Contiki instance?

fabijas, 2018-09-07:
Yes