contiki-os/contiki #2512: Telnet port??

misoehj, 2018-09-06:
I have a Zolertia Orion router. I programmed the binary files on the router for connectivity to the Internet over 6lbr. I created a 6lowpan network. But I still don’t see telnet services on the router
Will the Contiki/apps/telnet examples fix this. If so, how and what do I do?

misoehj, 2018-09-07:
Where will I find more information about how to do this?

erfanlinman, 2018-09-07:
You could look at this [document](https://github.com/contiki-os/contiki/wiki/Zolertia-z1-motes#contiki-examples-and-guides) and this [book](http://wireless.ictp.it/school_2015/book/).

misoehj, 2018-09-08:
I have a zoul remote (not z1). Is there any documentation for that? There are certain things that the z1 can do and zoul can't (like use Cooja).

misoehj, 2018-09-08:
And I don't see how this helps with my telnet problem.
I tried using the following commands (under contiki/examples/telnet-server) :
sudo make telnet-server.upload
sudo make TARGET=zoul BOARD=remote-revb telnet-server.upload
I get this error message:
using saved target 'zoul'
  CC        ../../cpu/cc2538/./ieee-addr.c
  CC        ../../cpu/cc2538/cc2538.lds
  CC        ../../cpu/cc2538/./startup-gcc.c
  CC        telnet-server.c
  LD        telnet-server.elf
obj_zoul/webserver-nogui.o: In function `webserver_log_file':
webserver-nogui.c:(.text.webserver_log_file+0x0): multiple definition of `webserver_log_file'
obj_zoul/shell-httpd.o:shell-httpd.c:(.text.webserver_log_file+0x0): first defined here
obj_zoul/webserver-nogui.o: In function `webserver_log':
webserver-nogui.c:(.text.webserver_log+0x0): multiple definition of `webserver_log'
obj_zoul/shell-httpd.o:shell-httpd.c:(.text.webserver_log+0x0): first defined here
obj_zoul/contiki-main.o: In function `main':
contiki-main.c:(.text.startup.main+0x168): undefined reference to `rime_driver'
obj_zoul/netstack.o: In function `netstack_init':
netstack.c:(.text.netstack_init+0x34): undefined reference to `rime_driver'
obj_zoul/nullsec.o: In function `input':
nullsec.c:(.text.input+0xc): undefined reference to `rime_driver'
collect2: error: ld returned 1 exit status
make: *** [telnet-server.elf] Error 1
rm obj_zoul/startup-gcc.o telnet-server.co

What does this mean?

erfanlinman, 2018-09-10:
Sorry, I used the wrong link; [here](https://github.com/contiki-os/contiki/tree/master/platform/zoul) it's the right one.
Try this one, it's comprehensive and very useful.
and error, it looks like it's about the wrong driver which cause some error; use the mentioned documentation.