contiki-os/contiki #2515: CC1310 Hardware timer problem

h1ppyq2, 2018-09-12:
Hello,

There is an issue, which we can not resolve. I use cycle while() with the GPT1(B) timer running, and check the value in the counter register to create timeout and exit the cycle. And then I put the value out with printf function. Some kind of this:

uint32_t time = 0;
F = 1;
while (F)
  { 
    if (TimerValueGet(GPT1_BASE, TIMER_B) > 30000) F = 0;
  }
time = TimerValueGet(GPT1_BASE, TIMER_B);
printf("\n\r Time: %u", time);

And the value that I get in the terminal is always increasing. After first measuring it is for example 30000, after second it is 33549, then 36544, etc. It always increases until 65535 with the step about 2.5-3k and then again from 30000. 
That is simple example with this problem. In my application this issue causes mistake in every 4-7 measurings using the hardware timer.
What is the problem?

alexrayne, 2018-09-20:
Can you provide more code? need to see context

greg-king5, 2018-09-23:
Your code must set the timer to zero before it waits.