contiki-os/contiki #2518: Cannot initiate 2 virtual nodes | Single 1 Tun Interface initiated

npcode15, 2018-09-25:
Hello,

I am trying to execute the rpl-udp example. I am trying to execute as native. Hence, I would like to have 2 virtual nodes. I am executing 
sudo ./udp-client.native 
sudo ./udp-server.native 

however, only first of the 2 initiates a tun interface whereas the other one fails.

Please let me know how to initiate 2 virtual nodes and successfully execute the example. 
Thank you.

npcode15, 2018-10-01:
@farcaller @heipei @debrouxl @mcr 

I apologise for the explicit tagging. It has been seven days since the issue was opened. Can someone please clarify the issue for me.

Thanks in advance.

mcr, 2018-10-01:
"failed to open tun device (you may be lacking permission)"

In the future, if you want help on a particular schedule, you are welcome to engage my consulting company.