contiki-os/contiki #2521: Compilation Error | IotLab-M3

npcode15, 2018-10-01:
Hello, 
I was trying to compile an application for the device iotlab-m3. The compilation fails with the following error:
```
../../Makefile.include:140: *** The target platform "iotlab-m3" does not exist (maybe it was misspelled?).  Stop.
```

I tried adding iotlab-m3 to the Makefile but didn't help.

The example I am trying to build is rpl-udp (cotnitki-ng) and udp-ipv6. Both gave the same error.

Also, I would like to know the recommended version of Contiki for development, contiki-ng or contiki3?

Thanks.

hieu1204, 2018-10-08:
The compile process is not simple like changing the target name and re-compile, you have to have a compatible contiki repo for a specific platform (e.g. ARM-M3). For the iotlab-m3 target, look at this tutorial, you can find the rpl-udp example on the iotlab-m3 platform in parts/contiki/example/ipv6/rpl-udp
https://www.iot-lab.info/tutorials/contiki-compilation/

hieu1204, 2018-10-08:
Note that the Contiki repo on the link above is modified to support the iotlab-m3 platform

npcode15, 2018-10-09:
@hieu1204 the tutorial also changed just the TARGET. What if I want to write the code locally and upload(flash) the compiled build? It should technically work. I have been doing the same for RIOT and I would like to replicate that with Contiki as well.

closed 2018-11-25 by npcode15