contiki-os/contiki #2528: Unable to overwrite file content

frank-freihube-rmcan-com, 2018-10-19:
I'm trying to overwrite parts of the file content (on the end of the file)

```c
hFile = EFS_open ("test.txt", CFS_WRITE);
if (hFile >= 0)
{
  result = EFS_write(hFile, "Hi1234", 6);
  EFS_close(hFile);
}

hFile = EFS_open ("test.txt", CFS_WRITE + CFS_APPEND);
if (hFile >= 0)
{
  result = EFS_seek(hFile, -4, CFS_SEEK_END); // result = 2
  result = EFS_write(hFile, "5678", 4);       // result = 4
  EFS_close(hFile);
}

hFile = EFS_open ("test.txt", CFS_READ);
if (hFile >= 0)
{
  result = EFS_seek(hFile, 0, CFS_SEEK_SET);  // just to be sure
  result = EFS_read(hFile, buf, 6);           // result is 6, buffer contains "Hi56\0\0"
  result = EFS_seek(hFile, 0, CFS_SEEK_END);  // result is 6 (test file length)
  EFS_close(hFile);
}
```

I did some other tests with and it seems the first 4 bytes are always as intended to be and the rest is \0
Is there a setting to be able to do overwriting of file content, is this function not supported at all or is this a bug?

MCU: STM32F417
Storage: 16MB NOR Flash on SPI

mhshd, 2019-04-28:
syntax of all the codes I saw was like this :
`hFile = EFS_open ("test.txt", CFS_WRITE | CFS_APPEND);`.

maybe this line provides some issues.

frank-freihube-rmcan-com, 2019-07-17:
As they are defined this way:
```c
#define CFS_WRITE 2
#define CFS_APPEND 4
```
There is no difference if you use `+` or `|` to combine them as they are only bit fields