contiki-os/contiki #2531: Trouble using Unicast and Multicast Together

npcode15, 2018-10-24:
Hello, 

Let me explain what I am trying to achieve first.

Assuming I have **x + 2 nodes:** 1 sender(client), 1 receiver (server) and x broadcasters (multicast nodes).

Client and Server exchange packets with each other whereas the rest of the nodes (x multicast nodes) multicast packets/messages to every other node including server and client. Each of the x nodes, multicasts to x+1 nodes (x-1 multicast nodes + server + client). 

**Device/Nodes:** iotlab-m3

**Tried Solutions:**

1. ```ipv6/simple-rpl-udp``` example 
**Problem:** service 190 not found | Multicast functioning was working just fine, unicast functioning had this error.

2. Extending ```udp-ipv6``` example to manually multicast to client and server only (using local DNS search).
**Problem:** Multicast application/code only searches for one hostname (but does that twice because I wrote the applicaiton to search for 2 hostnames - one client and one server)

3. Combining the ```udp-ipv6``` example with ```ipv6/multicast``` example. Using code for ```ipv6/multicast/root.c``` as it is. Merging the exisiting code from ```ipv6/multicast/sink.c``` with ```udp-ipv6/udp-server.c``` and ```udp-ipv6/udp-client.c```.

**Problem:** Undefined Reference to join_uip.mcast (something similar, I didn't copy the error).

**NOTE:** If needed I can provide the code for Solutions 2 and Solution 3. I can also provide exact errors/screenshots. I did provide them because i did not copy errors or screenshot them. Though I can reproduce them.

Please help me out with this issue thanks.

Thank you.

farcaller, 2018-10-29:
> @farcaller (sorry for explicit tagging)

you won't be excused for tagging people that have nothing to do with your question, no.

npcode15, 2018-10-29:
I apologize, I wasn't sure whom to tag (except Michael Richarson).

rahav, 2021-08-16:
@npcode15 Was this ever answered. I'm running into similar challenges. I tried to extend examples/multicast to send UDP messages. Then tried to have rpl-udp send multicast messages. Neither worked.

npcode15, 2021-11-03:
@rahav I don't think so.

closed 2021-11-03 by npcode15