contiki-os/contiki #2533: changing transmission range of an individual mote

abhiverma866, 2018-10-27:
I am trying to change transmission range of a particular node, but as a result I see transmission range of all the motes in the topology getting changed (which i don't want). Is there any way in contiki 3.0 or contiki-ng to achieve this task.

hieu1204, 2018-10-28:
You can change the Radio medium in cooja simulator (UDGM by default) to DGRM. To do so:
File -> New simulation -> Select DGRM in in Radio medium task -> Create, then add each link manually.

fjmsouza, 2023-07-01:
> You can change the Radio medium in cooja simulator (UDGM by default) to DGRM. To do so: File -> New simulation -> Select DGRM in in Radio medium task -> Create, then add each link manually.

Sincerely, after this I didn't find how change the range of every mote. There is other way? Through .csc, for example?