contiki-os/contiki #2548: Zolertia Zoul's dht22 sensor code is not working

Akash25595, 2018-12-28:
Hi, 
I Have tried to run the dht22.c code file, but after running it i am getting the error as "Failed to resd the sensor" everytime.
Could anyone please help me out.

closed 2018-12-31 by Akash25595

dserbu, 2019-03-16:
Did you found a solution?

Akash25595, 2019-03-20:
Hi dserbu,

please make the following changes as mentioned:

In DHT22.c: 
1. Intialising last_state = 0xFF;
2. Commenting on first assignment for last_state in read()  
   function and keeping second last_state as it is.
3. Change the type of variable inside the "int dht22_read_all
   ()" function from int to int16_t

In DHT22.h:
1. Change #define DHT22_READY_TIME from 20 to 40 value
2. Change the type of variable inside the "int dht22_read_all
   ()" function from int to int16_t

After making above changes, you will able to see the proper output from DHT22 sensor.
Let me know of you need my further help.