contiki-os/contiki #2554: Which Version of contiki should I use？

BigBigWolf-AI, 2019-01-16:
Hello, 
Recently，I found that a lot  of people choose to use contiki 2.7 instead of contiki 3.0, I was curious about this. For general experience，the newer the better, why they don't use  contiki 3.0, is it because of contiki 3.0 require more RAM of MCU or the code of contiki 3.0 is bigger than contiki 2.7 ?
truely hope someone could tell me about that, Thank You!

ekawahyu, 2019-01-17:
There are a ton of changes actually :), but mainly, as you already have said, the stack is growing. Especially for the IPv6 one. It depends on what you wanna build really. What is you target platform by the way? Are you gonna implement RIME or IPv6?

BigBigWolf-AI, 2019-01-18:
My platform is STM8L151C8T6 whose RAM is  4KB and ROM is 64KB, I only implement RIME and the wireless chip is the popular LoRa chip SX1278. So which version should I choose, contiki 2.7 or contiki 3.x ?
Thanks for answer my question.

ekawahyu, 2019-01-19:
Porting LoRa over Contiki OS from scratch? Which compiler? SDCC?

BigBigWolf-AI, 2019-01-27:
> Porting LoRa over Contiki OS from scratch? Which compiler? SDCC?

IAR

slcmcu, 2019-02-25:
用哪个版本都可以吧

ekawahyu, 2019-02-25:
Any version can be used, cause you basically have to create the IAR projects by yourself. It is not supported by default in contiki

BigBigWolf-AI, 2019-02-25:
我是担心3.x版本运行时占用的内存太大了，没办法在STM8L151C8T6上运行 @slcmcu @ekawahyu

BigBigWolf-AI, 2019-02-25:
中文直接答，牛逼呀 23333 @slcmcu

ekawahyu, 2019-02-25:
@BigBigWolf-AI If you stick with RIME, that STM8 has plenty. It is not sufficient if you want to use IPv6.

BigBigWolf-AI, 2019-02-25:
Thank You！ @ekawahyu

slcmcu, 2019-02-27:
英语不好,没办法只能说中文了,如果看不懂的话,自己找个翻译软件翻译一下吧

slcmcu, 2019-02-27:
占用内存的问题,最好实际测试一下,毕竟两个版本差别不是很大，就算是移植很不是很难。

BigBigWolf-AI, 2019-02-27:
谢谢 @slcmcu

slcmcu, 2019-02-27:
如果想使用IAR的话,就需要移植了,不过在linux编译如果可以的话就不用了

BigBigWolf-AI, 2019-02-27:
不用IAR, IAR真的是个大坑，不同IAR版本之间创建的代码不兼容。一个博主的移植文章，下载下来他的代码，运行，WTF，70多个错，直接放弃，现在在用Keil，超级好用

slcmcu, 2019-02-27:
@BigBigWolf-AI  你现在也在做contiki开发吗,用的是什么芯片

slcmcu, 2019-02-27:
如果GCC支持stm8的话,最好在linux下开发,这样方便升级和维护。

BigBigWolf-AI, 2019-02-27:
用的STM32, 目前并不知道如何在Linux开发STM32,有比较好的教程或者工具吗？
@slcmcu

BigBigWolf-AI, 2019-02-27:
以后会用STM8或者STM32L系列，主要是为了低功耗
@slcmcu

ekawahyu, 2019-02-27:
Any 8-bit MCU like 8051 or STM8 is pain to develop, especially since no GCC support. I would recommend STM32 family, or ARM in general.

BigBigWolf-AI, 2019-02-27:
Got it, thanks. But the STM8 can save a lot of money for company, so, even pain also have to try! @ekawahyu

ekawahyu, 2019-02-27:
Which exact part number of STM8 is it?