contiki-os/contiki #2565: How to read contents of file using CFS and print output of mote?

saideepd, 2019-02-25:
We are trying to read the contents of text file (or any other) using CFS and print the contents of the file in Mote output of COOJA Simulator. We don't want to read from or write to flash memory of mote. We just want to simulate it on simulator.
This is our code part:
```
int fd, r;
char buffer[BUFFER_SIZE];
char *filename = "text";
   
fd = cfs_open(filename, CFS_READ | CFS_WRITE);
printf("fd: %x\n",fd);
printf("File opened successfully.\n\n");
if(fd >= 0) {
      cfs_seek(rfd, 0, CFS_SEEK_END);
      r = cfs_read(fd, buffer, sizeof(buffer));
      printf("Read result: %d\n", r);
      printf("Data: %s\n", buffer);
}else {
      printf("ERROR: could not read from file.\n");
}
cfs_close(fd);
```

The issue is that the contents of file are not being read and printed on the mote output. The **fd** variable returns **0**. The print statement prints **Data:** but nothing after that. 
The **text** file exists in the same location as the code for reading file. Nothing is being read or written in buffer.

If anyone knows how to solve this issue please help.

smntgl, 2022-02-10:
Hi! Did you manage to find a solution for this problem?

nvt, 2022-02-10:
The statement <code>cfs_seek(rfc, 0, CFS_SEEK_END)</code> sets the offset to the end of the file. When a <code>cfs_read()</code> call is made on the file descriptor after that, it is expected to return 0.

closed 2022-02-10 by nvt

smntgl, 2022-02-10:
Actually I'm using this: 

`PROCESS_BEGIN();

  /* Open a file. */
  fd = cfs_open("measure", CFS_READ);
  if(fd >= 0) {
    cfs_seek(fd, 0, CFS_SEEK_SET);
    cfs_read(fd, buf, sizeof(buf));
  }
  else{
    printf("Cannot read the file\n")
  }`

In COOJA simulator I see a "Cannot read the file" in the mote output window. But actually if I run the code using the terminal and MAKE it works. Why in COOJA it doesn't? I put the file in the same directory of the .c source

saideepd, 2022-02-11:
Hey @smntgl Unfortunately I was not able to get things to work & had to drop the idea of using Cooja Simulator as there were multiple issues & roadblocks to tackle with Cooja in the short time I had at that time.

smntgl, 2022-02-11:
Thank you @saideepd. Unfortunately I can't drop the idea because I have an exam on Contiki and cooja at the university :_ ). I just need to simulate a noise sensor attached to a device running ContikiOs. I have some readings on a file and I don't know how to make the device read that file (simulating the sensor output). I don't know if you have some ideas since you gave up with cooja, but if you can help me I would appreciate it. Anyway thank you for the response.