contiki-os/contiki #2573: Math.h functions don't work.

abhiverma866, 2019-04-09:
Math.h functions don't work. Moreover, the variable with double is not showing anything. I want to perform exponentiation.
My sample code is:

PROCESS_THREAD(hello_world_process, ev, data)
{
  PROCESS_BEGIN();

  printf("Hello, world\n");
  printf("%f\n", pow(10,2.1));

  PROCESS_END();
}
Output log:

00:01.242    ID:1    Rime started with address 193.12.0.0.0.0.0.1
00:01.249    ID:1    MAC c1:0c:00:00:00:00:00:01 Contiki 3.0 started. Node id is set to 1.
00:01.254    ID:1    CSMA ContikiMAC, channel check rate 8 Hz, radio channel 26
00:01.262    ID:1    Tentative link-local IPv6 address fe80:0000:0000:0000:c30c:0000:0000:0001
00:01.264    ID:1    Starting 'Hello world process'
00:01.265    ID:1    Hello, world
00:01.265    ID:1    %f

abhiverma866, 2019-04-09:
The solution i found is that either to use fix point arithmetic or to type caste the output in order to get digits before decimal point.

niziak, 2019-04-09:
Try to Google about linker flags and `nano.specs`.

17dr000345, 2019-11-07:
use powf()