contiki-os/contiki #2585: tsch:! not associated, drop outgoing packet. (only channel scanning is happening)

souhilak, 2019-05-01:
Hi, when i try to run rpl-tsch in contiki this problem appears 
tsch:! not associated, drop outgoing packet.
When i creat a root-node (role-6dr)and source -node (role_6ln) the simulation work fine. 
but when i try to add other source node or intermediate nodes the problem of tsch not associated appears again !!!!!

Please help me

lamnd09, 2019-05-02:
Hi, what kind of scheduling strategy you use ? maybe you need to enable it by changing number from 0 to 1 in project-conf.h file. It will work.
/* Set to run orchestra /
#ifndef WITH_ORCHESTRA
#define WITH_ORCHESTRA 1
#endif / WITH_ORCHESTRA */

Anyway, if it does not work, could you show some screenshot of log ?

souhilak, 2019-05-02:
Hi @lamnd09
 i enabled orchestra by changing number to 1 in project-conf.h!!
Contact me in souhila.kettouche@gmail.com please

souhilak, 2019-05-03:
@lamnd09
[image]
[image]

lamnd09, 2019-05-04:
It seems like you have configured for the topology, which node would be root node.

closed 2019-05-04 by souhilak