contiki-os/contiki #2588: RPL-TSCH

souhilak, 2019-05-05:
Hi @simonduq @Conrad2210 @yatch
i'm new in contiki especially in TSCH protocol,
i tested RPL-TSCH-UDP by sending a trafic, a network that is formed by a grid of 16 nodes has been defined, in which only one of them takes on the role of coordinator , one sender and 14 intermediate nodes , but with basic rpl-udp gives me better results than with tsch!!
[image]
[image]
[image]

yatch, 2019-05-08:
Since I'm not active on contiki, I don't remember what is implemented in the example... I believe you need to learn what TSCH is, and how different TSCH is from CSMA.

>  with basic rpl-udp gives me better results than with tsch!!

Then, you will understand why you had better results in CSMA.

souhilak, 2019-05-09:
Thenkyou @yatch

closed 2019-05-13 by souhilak