contiki-os/contiki #2591: .'text' will not fit in region 'rom'  region 'rom' overflowed by xx bytes

OnurKlnc, 2019-05-22:
I am trying to use the example provided with contiki but i am getting this error can you guys please help me with this

[image]

/examples/er-rest-example/er-example-server.c

hknaydin, 2019-05-22:
You must update msp430 gcc 4.7.3

22 May 2019 Çar 12:05 tarihinde OnurKlnc <notifications@github.com> şunu
yazdı:

OnurKlnc, 2019-05-22:
Can you tell me how please

OnurKlnc, 2019-05-22:
> […](#)

Can you please explain how I can do that

hknaydin, 2019-05-22:
https://yazilimdnyasi.wordpress.com/2018/09/11/coojada-gcc-4-7-2-derleyicisi-ekleme/

OnurKlnc <notifications@github.com>, 22 May 2019 Çar, 12:45 tarihinde şunu
yazdı:

hknaydin, 2019-05-22:
http://ubuntuandsecurity.blogspot.com/2014/04/compiling-and-building-your-own-latest.html

OnurKlnc <notifications@github.com>, 22 May 2019 Çar, 12:45 tarihinde şunu
yazdı: