contiki-os/contiki #2595: Unable to update submodule

harshanavkis, 2019-06-10:
Whenever I run,
```
git submodule update --init
```

I get the following error:

```
error: Server does not allow request for unadvertised object        db648e9ea714946d05a94ccc4b8602a487f503d2

Fetched in submodule path '../../platform/stm32nucleo-spirit1/stm32cube-lib', but it did not contain db648e9ea714946d05a94ccc4b8602a487f503d2. Direct fetching of that commit failed.
```

Any idea how to fix this?

sillvios, 2023-03-23:
have you solved this?

sonali-05, 2023-03-23:
Hello Everyone
i want the routing table to be printed in the mote output and i am using
contiki 3.0. can anyone help on ths?

sonali-05, 2023-03-30:
Hello Everyone
Can you please guide me in printing number of hops between source and
destination.