contiki-os/contiki #2597: which library to use for i2c port in CC1310?

Lavr18, 2019-06-13:
Hi,

I am planning to connect a sensor to my CC1310 via i2c port.
Which library should I use? I didn't find any that is specific to CC13xx.
The only one I found is board-i2c.h for sensortag. Can I use it for CC13xx?

Best,
Aliaksei

rajeev1986, 2019-08-21:
Hi @Lavr18 . Yes, that works for CC1310 as long as you configure the ports correctly. I am using it with my custom design CC1310 based platform and it works great.

Sachingit1389, 2020-12-30:
Hello, 
I just want read the data with the i2c protocol sensor. I am trying to figure out from where the pin numbers in this code to be used for scl and sda. to read the sensor data. Can you help @rajeev1986 , @Lavr18 . I am using a launchpadxl cc1310

rajeev1986, 2021-01-11:
@Sachingit1389 
The I2C pin configs CC1310 LaumchPad are defined in platform/srf06-cc26xx/launchpad/cc1310/board.h
`#define BOARD_IOID_SCL            IOID_4`
`#define BOARD_IOID_SDA            IOID_5`

Change the above pins 4 and 5 to the I2c pins of your driver.
Hopefully, this helps.

Sachingit1389, 2021-01-22:
@rajeev1986 Thank you for your reply . Basically i want to read a sensor on i2c pins . Where i can find the supporting .c  file for this as source code . for reading the values of the sensor

rajeev1986, 2021-01-22:
@Sachingit1389,
Since launchpad does not come with onboard sensors, you will have to use the i2c sensor drivers of the sensortag located [here](https://github.com/contiki-os/contiki/tree/master/platform/srf06-cc26xx/sensortag).
You will see the source files and the header files (.c & .h) of the various sensors on the sensortag in this folder.
First, try to go over the drivers to see how they are structured before writing your own.

Since the contiki-os is now not being updated, any related questions should be now directed to [contiki-ng](https://github.com/contiki-ng/contiki-ng) repo. Hope this helps.