contiki-os/contiki #2598: why are DIOs not transmitted to all nodes?

hlsc1983, 2019-06-15:
I changed DEBUG_NONE to DEBUG_PRINT in rpl_imcp6.c in order to see this message "Received a DIO from ". What i have observed is that udp_senders within the transmission range of udp_sink receive DIO from the udp_sink since i can see the " Received DIO from  (sink ip address)" in the mote output window . However udp_senders outside the transmission range do not receive any DIO. Therefore, if they dont receive any DIO , then  how is DAG construction possible? MY understanding of DAG is that DIOs gets transmitted from the root to all the nodes. Please explain.

abhiverma866, 2019-07-31:
post you simulation serial log

hlsc1983, 2019-09-22:
very sorry for the delay. i didn't quite get what you are asking. you wanted to see the mote output?

abhiverma866, 2019-09-22:
yes.