contiki-os/contiki #2602: SELECTING AN OBJECTIVE FUNCTION FOR A SIMULATION ( OF0 - MRHOF)

JamesUribe, 2019-06-26:
Hi Everyone,

My name is Augusto and i want to ask your help kindly with this issue;

At this moment i am working in a thesis based on RPL protocol.

Now I want to compare differents perfomance of the OF (MRHOF and OF0) and maybe to use or design a new object function.

but i dont know how can i select or choose an objective funcion for my simulation ?

what file or option i need to modify for this purpose ?

I am using rpl- collect located at /contiki-3.0/examples/ipv6/rpl-collect.

Thanks,

hlsc1983, 2019-06-27:
Open Makefile.rpl and change rpl-of0.c to rplmrhof.c whenever necessary

JamesUribe, 2019-06-29:
Hi,

Thanks for your response.. im gonna try it.

JamesUribe, 2019-10-18:
HI

I wanna to say you thanks again for your response, and sorry for delay.

I have been working in my thesis, so i looked for the file called (Makefile.RPL)
as you said me but i couldn´t found it, i checked in the folder core/net/Rpl.

Or is located this file (Makefile.RPL)in each project´s folder (like /ipv6/examples/simple-udp) ?
where can i found it ?

Meanwhile for tests i have used the OF MHROF, which is the default OF in contiki-3.0

Thanks again

kamranzahoor, 2019-11-25:
@jamesuribe 
were you able to make changes in default objective functions? i too want to design new OF, can you help please?

JamesUribe, 2019-11-27:
Hi kamranzahoor, i used this paper [https://www.researchgate.net/publication/304572240](url), when it explains how you can modify and test with diferents OF, but still i couldn´t change or to get results, because i can´t find the file called Makefile.rpl which is mentioned in the paper.

kamranzahoor, 2019-11-27:
Thank you so much for your reply. But I am unable to open the link you have
send. Kindly resend the link or the paper directly.
Thanks once again. Regards.

JamesUribe, 2019-11-29:
[CoojaSimulatorManual11 (2).pdf](https://github.com/contiki-os/contiki/files/3903265/CoojaSimulatorManual11.2.pdf)

JamesUribe, 2019-11-29:
Hi kamranzahoor ,

I attached the file, where you can find a section called " Adjusting Objective Functions", i´ve tried to used it but, how i said in a previous message, i cant find the file Makefile.rpl, located at /home/user/contiki/core/net/rpl.

I will keep for your comments, 

Thanks

kamranzahoor, 2019-12-02:
thanks a lot.

kamranzahoor, 2019-12-10:
thanks a lot for your help, want to convey that "makefile.rpl" is present in contiki2.7, I managed to make changes. thanks again

kamranzahoor, 2019-12-10:
contiki2.7/core/net/rpl/makefile.rpl

JamesUribe, 2019-12-11:
Hi kamranzahoor ;

Thanks for your response, i am using contiki 3-0, and still  i can´t find the file "makefile.rpl" at the folder core/net/rpl ( i attached an image with the files included in this folder ). I have a question,  is it just create the file inside the folder or is it necesary other step ?

JamesUribe, 2019-12-11:
[image]

kamranzahoor, 2019-12-14:
it is present in contiki2.7 only, try your work on this

Liuyuwen-Zoe, 2020-03-12:
@JamesUribe  Do you have a solution at this 3.0 features do change OF

kamranzahoor, 2020-03-13:
i am not exactly sure but in rpl conf.h it can be changed at OCP(object
code point) where you can mention either of0 or MRHOF

kamranzahoor, 2020-03-13:
go through this
website:file:///H:/contikiplugins/contiki%20-%20Set%20paramaters%20of%20Cooja%20Simulation%20-%20Stack%20Overflow.html

JamesUribe, 2020-04-02:
Hi everybody,

First i want to say sorry for delay in my response.. but you know we are in quarantine at this time in argentina so its a little complicated.

About this file in contiki 3.0, i couldn´t found it in contiki..but i was searching in internet and i discovered this page

https://gitlab.utc.fr/tmaillar/SR04/commit/8bb4860138e31c4acd967d9bb3cff799f2f47fcf?view=parallel

there, you can select a link for download the Makefile.rpl.. still i have not tried.... but this weekend i am gonna do it..

i will wait for your comments,

James

JamesUribe, 2020-04-02:
> thanks a lot for your help, want to convey that "makefile.rpl" is present in contiki2.7, I managed to make changes. thanks again

> I managed to make changes. thanks again

Please,

Hi would you be so pleasant to indicate us, how did you do it ? what changes did you make ?? 

could you test the results ?

Thanks again, i hope your response.

kamranzahoor, 2020-04-02:
in makefile.rpl you can switch between two default objective functions i.e
of0 and mrhof

JamesUribe, 2020-05-22:
Hi everybody

First i want to say thanks to everybody, for your responses and attention.

Finally i could set the OF in contiki 3.0, using a file that i posted.

Now i am doing tests and my goal is design a new OF and can add it to rpl folder.

Thanks again

sk3110sk, 2020-07-04:
Hi James

In Contiki 3.0 have you modified only the rpl-conf.h file to change the OF?

mary-hada, 2020-11-11:
Hi
I'm working in develop objective function by take another routing metrics when calculate the rank
There is someone  work like me? I have some questions please help

JamesUribe, 2020-11-23:
Hi sk3110sk, just remember that you need to modify two files: Makefile.rpl and rpl-conf.h.

JamesUribe, 2020-11-23:
Hi Mary-hada, in my case only i changed the OF default what is being used by RPL, ( MRHOF or OF0), i did not modify the routing metricos or another parameter.

kamranzahoor, 2020-11-23:
Hi James where is makefile.rpl in contiki 3.0?? Please reply

mary-hada, 2020-12-08:
Please James can I talk with you? Please I need help

Meham26, 2021-03-22:
> Open Makefile.rpl and change rpl-of0.c to rplmrhof.c whenever necessary

it doesnt change a thing. see through the compilation files , there is no of0 file compiled..

Meham26, 2021-03-22:
change in make file of contiki/core/net/rpl folder as well as in the rpl-conf.h file of same folder . in rpl-conf.h file , u will find "#define RPL_OF rpl_of0 " just change it to rpl_mrhof and save.. while selecting mote, take it from the contiki folder not contiki 2.7

Chandranathkapat, 2023-06-04:
I'm trying to improve RPL objective function by implementing some new designed metrics in RPL using Contiki/cooja.
Can you please explain the changes to be made in brief?

poornimagowda, 2024-01-30:
Hi everyone,
I want to apply different optimization routing algorithms in Cooja. Can you please give me suggestions where to add and how to add the new algorithm?

kamranzahoor, 2024-01-30:
If you are using MRHOF OBJECTIVE FUNCTION  then routing can be enabled in
the parent selection part.

poornimagowda, 2024-01-30:
Thank you for your response. Can you please send any examples regarding that parent selection algorithm? I don't know about Cooja. I am currently doing research on this.

hknaydin, 2024-02-03:
First, you should create your own file, then select the of you created on the conf file. OF is selected as mrhof by default. You can easily do what you want by checking the steps here.

kamranzahoor, 2024-02-03:
Please Hakan can you mention those steps in programming portion, that will
be greatly appreciated

fuad686337, 2024-09-14:
Hello everyone
I am also trying to work on metrics to choose the best path in the Internet of Things network
As well as modifying the  MRHOF or OF0
But unfortunately, I have not started with that, I am trying to get more information about this work

fuad686337, 2024-09-16:
That is, in this work you only changed the default settings, and excuse me for this. What is the research contribution in this work that you did?
Can I be contacted via email fuad403273@quni.edu.ye