contiki-os/contiki #2610: Problem with Dense Networks

Rjlmota, 2019-08-14:
Hello, I'm trying to simulate some relatively dense IoT Networks using Cooja. But, I'm having trouble with the border router and the simulation speed.

I can only seem to get 40 routes at maximum with the rpl-border-router. When I change this line:
(contiki/platform/wismote/contiki-conf.h):
``
#define UIP_CONF_MAX_ROUTES   50
``
to 50 or higher, the simulation speed drops to 0% instantly after I start tunslip.

Besides, the area of deployment seem to have high impact, also. When I simulate in a 100m² area, I get 0 routes. The same setup, with 1000m² gives me 40 routes, though.

I would like to have some advise in how to simulate this dense networks, if possible.

kaleemullah360, 2019-08-20:
this might not be a good answer but.

based on my experience, simulation speed start decreasing over the time even at small network of three motes. this could be due to low system resources i.e CPU, RAM, etc.

my suggestions are: instead of using instant contiki os that runs over virtual machine use native installation on cent os,

or you can tweak cooja simulator java configuration etc.

Rjlmota, 2019-08-21:
Thanks for the response.
I was already doing that, but I still have the same problem.

I reduced the number of nodes and reduced the range of transmission. It seems to work now, but it isn't really what I wanted