contiki-os/contiki #2612: Someone destroyed the Wiki

nicolashimmelmann, 2019-09-03:
Hi,

I just visited the Wiki and it seems like the user "appleiinc" has overwritten all the pages in there with some acronyms and random letters.

I reverted those changes on all pages possible, however for some pages there was no earlier revision to revert to.

Cheers,
Nicolas

systemplado, 2019-09-04:
Hi,

It was once again deleted and there is no way of reverting them, I think.
[image] The person involved also opened an [issue](https://github.com/contiki-os/contiki/issues/2613).

Back to the problem at hand, are there any more ways to revert the Wiki? Or maybe even have a look or a copy of it? The waybackmachine (e.g. for [the arduino family](https://web.archive.org/web/20170716200716/https://github.com/contiki-os/contiki/wiki/The-Arduino-family)) might help but are there any other ways?

dlech, 2019-09-04:
Someone with admin access to this repo would be able to reset and force push the wiki via git.

https://help.github.com/en/articles/adding-or-editing-wiki-pages#cloning-wikis-to-your-computer

leosteinfeld, 2019-09-04:
I could get a particular page thanks to the google cache ("An introduction to Cooja",  we refer to it for an undergraduate course). Despite of contiki-ng, it would be nice to keep wiki pages and the repo as references.
Best
Leo S.

oliverschmidt, 2019-09-04:
Hi,

Given the current overall situation (number of Wiki contributions in the recent past, existence of contiki-ng) it seems reasonable to make the Wiki R/O for the public - which prevents things like we experienced now. It turns out that I have the right to do so - so I did it:

[image]

Hoping "everybody" agrees with that approach the next question is how to restore the Wiki again - and this time for good. I personally don't know how - and to be honest I don't have the time budget to investigate how to do it.

However, in case it turns out that I have the right to give somebody the necessary access rights who is willing and able to do it please follow up here and I'll provide the access rights.

If it turns out that I don't have the right to give the necessary access rights, I'm willing to reach out to someone who has and ask him to provide the rights.

Oliver

DanielFa, 2019-09-21:
As @dlech mentioned, there is a repository for the wiki changes, which you can revert, I already tested that.

The most recent commit, before any changes (as far as I could tell) was 8a707cdc8 (in the wiki).

You can clone it by cloning `git@github.com:contiki-os/contiki.wiki.git`

Revert it with `git revert --no-commit 8a707cdc8..HEAD`

And then commit it again. Using this, no history is destroyed (even if the history ist just deleting files). 
You should also be able to push it afterwards, I sadly do not have the permission for that. (and I actually do not know which permissions you would need for pushing to the wiki directly)

bthebaudeau, 2019-09-23:
I've just reverted the offending commit range.

closed 2019-09-23 by bthebaudeau