contiki-os/contiki #2618: GUI crashed after click on start in cooja simulator

vishal623, 2019-09-30:
Hello,

I have used cooja mote to create a network using UDP sender and sink. When I start the simulation it gives me the following error:

     [java] # A fatal error has been detected by the Java Runtime Environment:
     [java] #
     [java] #  SIGSEGV (0xb) at pc=0x0000000000000000, pid=14898, tid=0x00007f0a59e7c700
     [java] #
     [java] # JRE version: OpenJDK Runtime Environment (8.0_222-b10) (build 1.8.0_222-8u222-b10-1ubuntu1~18.04.1-b10)
     [java] # Java VM: OpenJDK 64-Bit Server VM (25.222-b10 interpreted mode linux-amd64 compressed oops)
     [java] # Problematic frame:
     [java] # C  0x0000000000000000
     [java] #
     [java] # Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
     [java] #
     [java] # An error report file with more information is saved as:
     [java] # /home/sneha/Contiki-OFFL-master/tools/cooja/build/hs_err_pid14898.log
     [java] Opened pcap file!
     [java] #
     [java] # If you would like to submit a bug report, please visit:
     [java] #   http://bugreport.java.com/bugreport/crash.jsp
     [java] # The crash happened outside the Java Virtual Machine in native code.
     [java] # See problematic frame for where to report the bug.
     [java] #
     [java] Java Result: 134

I have already set the java_home environment variable. I want to know is it related to JAVA or related to code?

Thanks,
Vishal

dpapdpap, 2020-05-29:
I have exactly the same problem.
In a clear installation of contiki-ng (WSL2, UBUNTU 20.4, openjdk-8-amd64) I try the Hello-world example. 
I create a cooja-node. It compiles without errors or warnings but, as soon as the start is pressed, the IDE closes by giving the following error messages :
[java] # A fatal error has been detected by the Java Runtime Environment:
     [java] #
     [java] #  SIGSEGV (0xb) at pc=0x00007f6719ecbf4d, pid=4215, tid=0x00007f66eae3a700
     [java] #
     [java] # JRE version: OpenJDK Runtime Environment (8.0_252-b09) (build 1.8.0_252-8u252-b09-1ubuntu1-b09)
     [java] # Java VM: OpenJDK 64-Bit Server VM (25.252-b09 mixed mode linux-amd64 compressed oops)
     [java] # Problematic frame:
     [java] # C  [libc.so.6+0x96f4d]
     [java] #
     [java] # Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
     [java] #
     [java] # An error report file with more information is saved as:
     [java] # /home/dpap/contiki-ng/tools/cooja/build/hs_err_pid4215.log
     [java] #
     [java] # If you would like to submit a bug report, please visit:
     [java] #   http://bugreport.java.com/bugreport/crash.jsp
     [java] # The crash happened outside the Java Virtual Machine in native code.
     [java] # See problematic frame for where to report the bug.
     [java] #
     [java] Java Result: 134

Did you find a solution to tell me? or can someone help us?
It's a shame to get there and not be able to continue!

tharindupr, 2021-02-08:
Same problem. Is there any fix for this issue?

YiHang-BaiLu, 2022-05-06:
I can run the simulation, but sometimes it crashes after a while by giving the following error like yours:
`     [java] #
     [java] # A fatal error has been detected by the Java Runtime Environment:
     [java] #
     [java] #  SIGSEGV (0xb) at pc=0x00007fb96ef6738c, pid=14627, tid=140434412836608
     [java] #
     [java] # JRE version: OpenJDK Runtime Environment (7.0_201) (build 1.7.0_201-b00)
     [java] # Java VM: OpenJDK 64-Bit Server VM (24.201-b00 mixed mode linux-amd64 compressed oops)
     [java] # Derivative: IcedTea 2.6.17
     [java] # Distribution: Ubuntu 14.04 LTS, package 7u211-2.6.17-0ubuntu0.1
     [java] # Problematic frame:
     [java] # C  [mtype789.cooja+0x1238c]  select_packet+0x12d
     [java] #
     [java] # Core dump written. Default location: /home/john/OST-master/tools/cooja/build/core or core.14627
     [java] #
     [java] # An error report file with more information is saved as:
     [java] # /home/john/OST-master/tools/cooja/build/hs_err_pid14627.log
     [java] #
     [java] # If you would like to submit a bug report, please include
     [java] # instructions on how to reproduce the bug and visit:
     [java] #   http://icedtea.classpath.org/bugzilla
     [java] # The crash happened outside the Java Virtual Machine in native code.
     [java] # See problematic frame for where to report the bug.
     [java] #
     [java] Java Result: 134
`
I think it related to code rather than system. I had tried two same network but different topo, and it crashed at different time.

YiHang-BaiLu, 2022-05-06:
I can run the simulation, but sometimes it crashes after a while by giving the following error like yours:
    `[java] #` 
    `[java] # A fatal error has been detected by the Java Runtime Environment:` 
     [java] #
     [java] #  SIGSEGV (0xb) at pc=0x00007fb96ef6738c, pid=14627, tid=140434412836608
     [java] #
     [java] # JRE version: OpenJDK Runtime Environment (7.0_201) (build 1.7.0_201-b00)
     [java] # Java VM: OpenJDK 64-Bit Server VM (24.201-b00 mixed mode linux-amd64 compressed oops)
     [java] # Derivative: IcedTea 2.6.17
     [java] # Distribution: Ubuntu 14.04 LTS, package 7u211-2.6.17-0ubuntu0.1
     [java] # Problematic frame:
     [java] # C  [mtype789.cooja+0x1238c]  select_packet+0x12d
     [java] #
     [java] # Core dump written. Default location: /home/john/OST-master/tools/cooja/build/core or core.14627
     [java] #
     [java] # An error report file with more information is saved as:
     [java] # /home/john/OST-master/tools/cooja/build/hs_err_pid14627.log
     [java] #
     [java] # If you would like to submit a bug report, please include
     [java] # instructions on how to reproduce the bug and visit:
     [java] #   http://icedtea.classpath.org/bugzilla
     [java] # The crash happened outside the Java Virtual Machine in native code.
     [java] # See problematic frame for where to report the bug.
     [java] #
     [java] Java Result: 134

I think it related to code rather than system. I had tried two same network but different topo, and it crashed at different time.

YiHang-BaiLu, 2022-05-06:
I can run the simulation, but sometimes it crashes after a while by giving the following error like yours:
```
[java] #
[java] # A fatal error has been detected by the Java Runtime Environment:
[java] #
[java] # SIGSEGV (0xb) at pc=0x00007fb96ef6738c, pid=14627, tid=140434412836608
[java] #
[java] # JRE version: OpenJDK Runtime Environment (7.0_201) (build 1.7.0_201-b00)
[java] # Java VM: OpenJDK 64-Bit Server VM (24.201-b00 mixed mode linux-amd64 compressed oops)
[java] # Derivative: IcedTea 2.6.17
[java] # Distribution: Ubuntu 14.04 LTS, package 7u211-2.6.17-0ubuntu0.1
[java] # Problematic frame:
[java] # C [mtype789.cooja+0x1238c] select_packet+0x12d
[java] #
[java] # Core dump written. Default location: /home/john/OST-master/tools/cooja/build/core or core.14627
[java] #
[java] # An error report file with more information is saved as:
[java] # /home/john/OST-master/tools/cooja/build/hs_err_pid14627.log
[java] #
[java] # If you would like to submit a bug report, please include
[java] # instructions on how to reproduce the bug and visit:
[java] # http://icedtea.classpath.org/bugzilla
[java] # The crash happened outside the Java Virtual Machine in native code.
[java] # See problematic frame for where to report the bug.
[java] #
[java] Java Result: 134
```
I think it related to code rather than system. I had tried two same network but different topo, and it crashed at different time.

tejasvi, 2022-05-06:
The above crash generally originates from the crash in Contiki which can occur due to invalid memory access (SIGSEGV), race conditions, etc. If there is no obvious source of error, start by logging the execution of OS entry points. If need arises, log the execution of each line of the relevant part of the OS. From there you can find which operation is causing the crash.

Keep in mind that Contiki's `PROCESS` etc. macros are based on [protothreads](https://en.wikipedia.org/wiki/Protothread). Local variables (including temporaries) in the protothread can not be trusted to persist their value on reaching next instruction. E.g. previously calculated index value might become out of bounds (or incorrect) when later indexing the array. Instead use global variables and function calls which themselves can use local variables without issues.