contiki-os/contiki #2627: XM-1000 mote

asanz060, 2019-12-04:
We are trying to use the XM-1000 motes with Contiki 3.0. In order to make sure Contiki works we added the Platform and Tools files that are available in the AdvanticSys web page, in the Resources section, following the instructions of the XM1000_README.txt.

However, when we try to upload the hello world program into the mote in order to check if it works, we do not get any error message but the program is not uploaded either. The result that we get when we do "make TARGET=xm1000 hello-world.upload" is the following:

[image]

We do not know where the problem is, as XM1000 and Contiki should be compatible. Does anybody know what we have to do to make it work?

Thank you!

gajrajkuldeep, 2020-01-24:
Hi, I am facing the same problem. If you have sorted out then please let us know the trick

ogulec, 2021-05-28:
any update on this?

hueberflorian, 2021-12-12:
Hello, if you want to upload a program to the XM-1000 mote, use the command:

`sudo make hello-world.upload TARGET=xm1000`

And if you want to run the program already on your mote (in this case print the hello world message), type:

`sudo make login TARGET=xm1000`