contiki-os/contiki #2643: JAVA_HOME environment variable not set, Cooja motes may not compile

dpapdpap, 2020-04-14:
I know that many people have asked about this problem but I did not find an answer anywhere, so I ask in turn.
Why contiki /cooja shows this problem when i execute "sudo ant run", since the variable environment $ JAVA_HOME is set (in .profile) and when I echo $ JAVA_HOME it gives me the correct value. How else can I determine it?
Please give me an answer because I'm stuck and as a beginner I'm on Linux I'm ready to give up.

ahmadmr81, 2020-04-14:
Hi

This is from my experience and understanding: Using “sudo” means you are switching to an admin-level account. JAVA_HOME must be set also for that account, along with all other JAVA environment settings.

My experience, running Cooja with an administrative level creates a lot of problems... Just use “ant run” or “ant run_bigmem”

Sincerely;
Ahmed Raoof
Ph.D. Candidate
Dep. of Systems & Computer Engineering
Carleton University
Ottawa, ON, Canada

[External Email]

This email contains links to content or websites. Always be cautious when clicking on external links or attachments. If in doubt, please forward suspicious emails to phishing@carleton.ca.
-----End of Disclaimer-----

dpapdpap, 2020-04-15:
Ahmed thank you for your quick and correct answer.
Indeed, the problem was "sudo". Now contiki 3 recognizes JAVA_HOME but I still can't compile the nodes. I get the error: gcc: error trying to exec 'cc1': execvp: No such file or directory, but I'll put it in another issue
thanks again

closed 2020-04-19 by dpapdpap