contiki-os/contiki #2646: Build Failed

Twinkle0057, 2020-04-18:
Buildfile: /home/twinkle/contiki/tools/cooja/build.xml

init:

compile:
    [mkdir] Created dir: /home/twinkle/contiki/tools/cooja/build
    [javac] Compiling 164 source files to /home/twinkle/contiki/tools/cooja/build
    [javac] /home/twinkle/contiki/tools/cooja/java/org/contikios/cooja/contikimote/interfaces/ContikiEEPROM.java:41: error: package javax.xml.bind does not exist
    [javac] import javax.xml.bind.DatatypeConverter;
    [javac]                      ^
    [javac] /home/twinkle/contiki/tools/cooja/java/org/contikios/cooja/contikimote/interfaces/ContikiEEPROM.java:281: error: cannot find symbol
    [javac]       element.setText(DatatypeConverter.printBase64Binary(getEEPROMData()));
    [javac]                       ^
    [javac]   symbol:   variable DatatypeConverter
    [javac]   location: class ContikiEEPROM
    [javac] /home/twinkle/contiki/tools/cooja/java/org/contikios/cooja/contikimote/interfaces/ContikiEEPROM.java:290: error: cannot find symbol
    [javac]           setEEPROMData(DatatypeConverter.parseBase64Binary(element.getText()));
    [javac]                         ^
    [javac]   symbol:   variable DatatypeConverter
    [javac]   location: class ContikiEEPROM
    [javac] Note: Some input files use or override a deprecated API.
    [javac] Note: Recompile with -Xlint:deprecation for details.
    [javac] Note: Some input files use unchecked or unsafe operations.
    [javac] Note: Recompile with -Xlint:unchecked for details.
    [javac] 3 errors

BUILD FAILED
/home/twinkle/contiki/tools/cooja/build.xml:77: Compile failed; see the compiler error output for details.

I got above error message
I use openjdk 13
I didn't get my issue clarified

mguc, 2020-04-18:
I would suggest that you start using the [Contiki-NG](https://github.com/contiki-ng/contiki-ng) repository. This one is deprecated and not maintained.

Twinkle0057, 2020-04-18:
Thank you man

closed 2020-04-18 by Twinkle0057