contiki-os/contiki #2653: Collect-view and Script Runner error

shalinystalin, 2020-06-27:
Script Runner and collect-view application in my cooja simulator is not working. 
Collect-view application is not showing any results. Script Runner is showing some errors. Can anyone help me to solve this issue.
Screenshot of Script Editor error page and collect-view application is attached here.

[image]

[image]

Libtissam, 2023-04-07:
Hello! I have the same problem. Do you solve it?

hknaydin, 2023-04-07:
Hello, I faced this problem before. I kept a blog below that writes how to
solve it.

https://yazilimdnyasi.wordpress.com/2019/10/19/collect-view-example-for-contiki-os/

Libtissam ***@***.***>, 7 Nis 2023 Cum, 15:01 tarihinde şunu
yazdı:

> Hello! I have the same problem. Do you solve it?

Libtissam, 2023-04-08:
@hknaydin, Thank you, i will see it