contiki-os/contiki #2662: Implementating exponential function

vishal623, 2020-09-30:
Hello,

I want to implement a sigmoid equation in the er-coap application. But the biggest challenge is the exponential function. 

I have found the msp430-gcc supports some expf() function but I did not find it in Contiki.

Can anyone guide me on how can I implement exp() in er-coap or any application?

atiselsts, 2020-09-30:
What exactly is your problem? Did you trying adding `#include <math.h>` and linking with libm to your application? For the latter, add `TARGET_LIBFILES += -lm` to the Makefile.

vishal623, 2020-09-30:
I did try this long back with contiki-3.0 but solution did not work. 

I am using msp430-gcc-4.6.3 toolchain. And I did not find any "libm.a" in the folder which include exp or expf function. 

Whether adding TARGET_LIBFILES += -lm in the Makefile, is it work?

vishal623, 2020-09-30:
And one more thing, I am planning to add it in the code of apps/er-coap/er-coap-transactions.c. Not in example folder.

vishal623, 2020-09-30:
I want to apply sigmoid function, so power function is also required. Is "^" works in Contiki?