contiki-os/contiki #2670: Cooja simulator problem

mary-hada, 2020-12-05:
Hi 
I'm trying to make simple network in cooja 
I have this problem 
Someone know what is the error 
please help me
[image]

cherif-benali, 2021-02-11:
is the button "create" activated? i mean after  clicking "compile" .
if you click "compile" and the button "create" is activated, i think  the red lines are just warrings

Caucorico, 2021-02-22:
@mary-hada,

you didn't install the gcc-msp430 package on your computer. Install it and try again the compilation.

The package can be found here for **debian** :
https://packages.debian.org/sid/gcc-msp430
You can probably install it with apt-get.

For **arch** distrib, you can find it in AUR :
https://aur.archlinux.org/packages/gcc-msp430

Rahi165, 2024-09-26:
did you get the solution?? I am facing the same problem..